lgsvl/simulator #142: Unity project does not work properly

PatrickSTAR001, 2019-04-23:
I am opening its unity project file but I have been reporting an error. I installed git lfs. Is it because the unity version is wrong? This project was developed with the 2018.2.4 version of unity, and I am using a higher version, but now Unity official website has no version of 2018.2.4

martins-mozeiko, 2019-04-23:
What error do you see exactly?
It may be that some thing have changed in newer version and you might need to adjust code, scene or prefabs. It is better if you use exact version we are using - 2018.2.4.

You can get 2018.2.4 version for Windows on this page: https://unity3d.com/get-unity/download/archive
For Linux it is available here: https://forum.unity.com/threads/unity-on-linux-release-notes-and-known-issues.350256/page-2#post-3603070

PatrickSTAR001, 2019-04-24:
@martins-mozeiko  thanks for your reply ,error CS026: The type or namespace name "WebSocketSharp" could not be found(are you missing a using directive or an assembly reference) 
and  Windows  page can not find the 2018.2.4 version,I will try the Linux one.I am very grateful to you

martins-mozeiko, 2019-04-24:
This means you did not have git lfs installed when you were cloning repository. We use git lfs for large file storage so cloning repository is more efficient. You need to install it, then run `git lfs install` command to enable necessary git hooks. After that try cloning repository again or just fetching missing files - this error will go way. Check the readme for more detailed steps how to set up git lfs.

PatrickSTAR001, 2019-04-25:
@martins-mozeiko I have solved this problem with your help, thank you very much.

closed 2019-04-26 by martins-mozeiko