lgsvl/simulator #144: How to change the lidar configuration

khaledelmadawi, 2019-04-28:
Hi All,
First I would like to thank you again for this outstanding framework, second how can I change the configuration of lidar sensor as shown in Lidar demo presented in youtube?

[image]

Thanks

khaledelmadawi, 2019-04-28:
I could figure out how it works, but I found that the maximum number of layers is 128, can it be extended in the coming releases?

martins-mozeiko, 2019-04-29:
This is only limitation in UI. You can change max number in [LidarSensor.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/LidarSensor/LidarSensor.cs#L25) file.
We put 128 there as it is pretty reasonable number. But even at 128 rays simulation will have very low performance. It requires extra optimizations that we'll need to do. Going higher than that will pretty unusable for real-time simulation.

khaledelmadawi, 2019-05-01:
Thank you martin for your answer, how do you recommend he applied optimizations would be?

martins-mozeiko, 2019-05-01:
First you need to measure performance to understand where time is spent. Is it more GPU or CPU. Then think about how to change to reduce it, maybe use multiple threads, maybe some extra native code to calculate result faster. Not sure yet.

martins-mozeiko, 2019-08-19:
With our new simulator [2019.07-rc1](https://github.com/lgsvl/simulator/releases/tag/2019.07-rc1) release we have greatly optimized lidar performance. It is now able to do 128 ray lidar simulation in real time. Try it out and if you see any other problems feel free to open a new issue.

closed 2019-08-19 by martins-mozeiko