lgsvl/simulator #145: Hdmap and 3D model related

Derderdi, 2019-04-30:
Hello, I just learned your LG simulator. I want to ask a few questions now.
1. if the LG simulator support generating 3D world model from HD map(such as opendrive). If it support this feature, how to do it ?
2.if there is no 3D model, LG could run or not?
I'm looking forward to your reply. thanks :)

martins-mozeiko, 2019-05-01:
Currently we do not have functionality to import existing map. The workflow is opposite - we annotate map in simulator, and then export Apollo or Autoware maps from simulator.

That said, we have on our roadmap feature to import opendrive map.

By 3d model you mean 3d model for car? If you do not have your own 3d model for car, you can use existing cars in simulator.

Derderdi, 2019-05-05:
not 3d model for car. I want to know if LG simulator support no render mode.
If no model rendering, can I run it? So I can make it run faster to get some results I want.

martins-mozeiko, 2019-05-06:
Currently this kind of option is not exposed in UI. But you can disable main camera rendering by disabling "Driver Camera" object for vehicle prefab. After you save prefab, you can build new binary and it will not render anything on screen - in this case all interaction will need to happen through API.

Derderdi, 2019-05-07:
Ok, thank you very much for solving my confusion. 
wish you all the best.

closed 2019-06-13 by martins-mozeiko