lgsvl/simulator #149: how to let the ego-vehicle's recognize traffic lights and stop properly with python api mode

fengsiyu, 2019-05-07:
Hey, LGSVL Teams:
       recently, I'm writting the code with python api and I want to let the ego-vehicle obey the traffic rules when the camera sensors detected the red light. Then I can write some codes to stop the ego-vehicle.
       So, I opened the camera sensor and lidar sensor in API and I want to konw how to detect the traffic light？ Saving the camera photo and call the yolo algorithm or other method. Beacause I want using the standlone mode to make experiments.
       Thank you for your answer.

martins-mozeiko, 2019-05-08:
Our Python API is not really meant to control EGO vehicle in real time.

Python API is more meant for setting up scenarios to run with simulator. Idea is to set up other vehicles, set up their waypoints, connect to AD stack with ROS, and then run simulation for N seconds to see if there will be any collisions.

If you really want to access camera image from API then check out [Api/quickstart/06-save-camera-image.py](
https://github.com/lgsvl/simulator/blob/master/Api/quickstart/06-save-camera-image.py) script. This script shows how to save camera image to file. Then your Python API could load image, do traffic light detection and decide on what to do.

If you want to control vehicle in real time I suggest to create ROS node to subscribe to sensor messages and provide control command back.

fengsiyu, 2019-05-08:
If I want to set up scenarios, such as the NPC Vehicle and run the simulation with Baidu Apollo, what should i do? I means  the sepcify steps in the correct order? 
      eg.  1. run the LGSVL simulator
             2. run the python script to set up scenarios
             3. run the bootstrap script
             4. run the rosbridge script
I don't konw what's the correct order, could you help me to solve this problem? I'm doing my experiments with your simulator and Baidu Apollo.
          Thank you for your timely help.

martins-mozeiko, 2019-05-08:
You can have Apollo with boostrap+rosbridge running in background all the time. You don't need to stop it when running different API scripts.

To launch simulator with API you need to do following:

1) launch simulator
2) launch python script
3) python script should load map, create vehicle(s) and call connect_bridge function to correct IP address
4) python script should also enable GPS/IMU sensors for localization and as necessary camera & lidar
5) then your script should wait to allow you set up route Apollo dreamview.
6) then you can call sim.run() to start simulation.

fengsiyu, 2019-05-09:
I'm extremely grateful to you for solving my problems. 
But  when I use python API to enable GpsSensors as the same with other sensors, the compiler shows '”AttributeError: module 'lgsvl' has no attribute 'GpsSensor“' 
my code is as below:
`for s in sensors:
    if isinstance(s, lgsvl.GpsSensor):
        s.enabled = True`
And I read the lgsvl package's __init__.py source code, then I found  the GpsSensors module haven't been imported. the source code as below:
`# This software contains code licensed as described in LICENSE.
#

from .geometry import Vector, BoundingBox, Transform
from .simulator import Simulator, RaycastHit, WeatherState
from .sensor import Sensor, CameraSensor, LidarSensor, ImuSensor
from .agent import AgentType, VehicleControl, AgentState, Vehicle, EgoVehicle, NpcVehicle, Pedestrian, DriveWaypoint, WalkWaypoint
` 
what should I do? Clicking the GPS options manually in the simulator or other method?

And another question is that when I run the bootstrap and rosbridge firstly, then I run the simlator and launch the python script. Then I open the Apollo Dreamview, I found it's impossible to get the map data and run the simulation as usual.

Beacause I don't understand the 

> step 5: your script should wait to allow you set up route Apollo dreamview.
How to write code to wait set up route Apollo dreamview? Could you give me an example code? 
I guess I can read the example code to  understand your answer.
Thank you for your kindly help.

martins-mozeiko, 2019-05-09:
This seems to be mistake in `init.py`. You can add GpsSensor to the list of other sensors and it will do the import correctly.

Map data will show up in Dreamview only when localization works. To get localization working you need to enable GPS and IMU sensors.

In step 5 I wanted to say that your script should pause and wait a bit to allow you to set route in Dreamview for car to drive. To set route you either do it manually in browser window of Dreamview. Or you need to find out what topic / ROS message to send to Apollo to set route. I'm not sure what topic you need to use, as simulator never had need to do this. You'll need to read Apollo code to understand this.

fengsiyu, 2019-05-09:
I modified the init.py and reinstall the lgsvl module, then the GpsSensor can work correctly. But I run these scripts in the order as below:
1. run the bootstrap and rosbridge
2. run the simulator and let it stay in the main window
3. launch the scenarios setting python script
when I launch the python script, a error occurs in the terminal and the simulator crashed:
`=================================================================
Got a SIGABRT while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================
Aborted
`

martins-mozeiko, 2019-05-09:
Can you send use logfile from Simulator?
If you are running binary build then it is at `~/unity3d/LG Silicon Valley Lab/Automotive Simulator/Player.log`.

fengsiyu, 2019-05-09:
`Desktop is 1920 x 1080 @ 120 Hz
Initialize engine version: 2018.2.4f1 (fe703c5165de)
GfxDevice: creating device client; threaded=1
Renderer: GeForce GTX 1060/PCIe/SSE2
Vendor:   NVIDIA Corporation
Version:  4.5.0 NVIDIA 390.87
GLES:     0
 GL_AMD_multi_draw_indirect GL_AMD_seamless_cubemap_per_texture GL_AMD_vertex_shader_viewport_index GL_AMD_vertex_shader_layer GL_ARB_arrays_of_arrays GL_ARB_base_instance GL_ARB_bindless_texture GL_ARB_blend_func_extended GL_ARB_buffer_storage GL_ARB_clear_buffer_object GL_ARB_clear_texture GL_ARB_clip_control GL_ARB_color_buffer_float GL_ARB_compressed_texture_pixel_storage GL_ARB_conservative_depth GL_ARB_compute_shader GL_ARB_compute_variable_group_size GL_ARB_conditional_render_inverted GL_ARB_copy_buffer GL_ARB_copy_image GL_ARB_cull_distance GL_ARB_debug_output GL_ARB_depth_buffer_float GL_ARB_depth_clamp GL_ARB_depth_texture GL_ARB_derivative_control GL_ARB_direct_state_access GL_ARB_draw_buffers GL_ARB_draw_buffers_blend GL_ARB_draw_indirect GL_ARB_draw_elements_base_vertex GL_ARB_draw_instanced GL_ARB_enhanced_layouts GL_ARB_ES2_compatibility GL_ARB_ES3_compatibility GL_ARB_ES3_1_compatibility GL_ARB_ES3_2_compatibility GL_ARB_explicit_attrib_location GL_ARB_explicit_uniform_location GL_ARB_fragmen
t_coord_conventions GL_ARB_fragment_layer_viewport GL_ARB_fragment_program GL_ARB_fragment_program_shadow GL_ARB_fragment_shader GL_ARB_fragment_shader_interlock GL_ARB_framebuffer_no_attachments GL_ARB_framebuffer_object GL_ARB_framebuffer_sRGB GL_ARB_geometry_shader4 GL_ARB_get_program_binary GL_ARB_get_texture_sub_image GL_ARB_gl_spirv GL_ARB_gpu_shader5 GL_ARB_gpu_shader_fp64 GL_ARB_gpu_shader_int64 GL_ARB_half_float_pixel GL_ARB_half_float_vertex GL_ARB_imaging GL_ARB_indirect_parameters GL_ARB_instanced_arrays GL_ARB_internalformat_query GL_ARB_internalformat_query2 GL_ARB_invalidate_subdata GL_ARB_map_buffer_alignment GL_ARB_map_buffer_range GL_ARB_multi_bind GL_ARB_multi_draw_indirect GL_ARB_multisample GL_ARB_multitexture GL_ARB_occlusion_query GL_ARB_occlusion_query2 GL_ARB_parallel_shader_compile GL_ARB_pipeline_statistics_query GL_ARB_pixel_buffer_object GL_ARB_point_parameters GL_ARB_point_sprite GL_ARB_polygon_offset_clamp GL_ARB_post_depth_coverage GL_ARB_program_interface_query GL_ARB_provoki
ng_vertex GL_ARB_query_buffer_object GL_ARB_robust_buffer_access_behavior GL_ARB_robustness GL_ARB_sample_locations GL_ARB_sample_shading GL_ARB_sampler_objects GL_ARB_seamless_cube_map GL_ARB_seamless_cubemap_per_texture GL_ARB_separate_shader_objects GL_ARB_shader_atomic_counter_ops GL_ARB_shader_atomic_counters GL_ARB_shader_ballot GL_ARB_shader_bit_encoding GL_ARB_shader_clock GL_ARB_shader_draw_parameters GL_ARB_shader_group_vote GL_ARB_shader_image_load_store GL_ARB_shader_image_size GL_ARB_shader_objects GL_ARB_shader_precision GL_ARB_shader_storage_buffer_object GL_ARB_shader_subroutine GL_ARB_shader_texture_image_samples GL_ARB_shader_texture_lod GL_ARB_shading_language_100 GL_ARB_shader_viewport_layer_array GL_ARB_shading_language_420pack GL_ARB_shading_language_include GL_ARB_shading_language_packing GL_ARB_shadow GL_ARB_sparse_buffer GL_ARB_sparse_texture GL_ARB_sparse_texture2 GL_ARB_sparse_texture_clamp GL_ARB_spirv_extensions GL_ARB_stencil_texturing GL_ARB_sync GL_ARB_tessellation_shader GL_A
RB_texture_barrier GL_ARB_texture_border_clamp GL_ARB_texture_buffer_object GL_ARB_texture_buffer_object_rgb32 GL_ARB_texture_buffer_range GL_ARB_texture_compression GL_ARB_texture_compression_bptc GL_ARB_texture_compression_rgtc GL_ARB_texture_cube_map GL_ARB_texture_cube_map_array GL_ARB_texture_env_add GL_ARB_texture_env_combine GL_ARB_texture_env_crossbar GL_ARB_texture_env_dot3 GL_ARB_texture_filter_anisotropic GL_ARB_texture_filter_minmax GL_ARB_texture_float GL_ARB_texture_gather GL_ARB_texture_mirror_clamp_to_edge GL_ARB_texture_mirrored_repeat GL_ARB_texture_multisample GL_ARB_texture_non_power_of_two GL_ARB_texture_query_levels GL_ARB_texture_query_lod GL_ARB_texture_rectangle GL_ARB_texture_rg GL_ARB_texture_rgb10_a2ui GL_ARB_texture_stencil8 GL_ARB_texture_storage GL_ARB_texture_storage_multisample GL_ARB_texture_swizzle GL_ARB_texture_view GL_ARB_timer_query GL_ARB_transform_feedback2 GL_ARB_transform_feedback3 GL_ARB_transform_feedback_instanced GL_ARB_transform_feedback_overflow_query GL_ARB_t
ranspose_matrix GL_ARB_uniform_buffer_object GL_ARB_vertex_array_bgra GL_ARB_vertex_array_object GL_ARB_vertex_attrib_64bit GL_ARB_vertex_attrib_binding GL_ARB_vertex_buffer_object GL_ARB_vertex_program GL_ARB_vertex_shader GL_ARB_vertex_type_10f_11f_11f_rev GL_ARB_vertex_type_2_10_10_10_rev GL_ARB_viewport_array GL_ARB_window_pos GL_ATI_draw_buffers GL_ATI_texture_float GL_ATI_texture_mirror_once GL_S3_s3tc GL_EXT_texture_env_add GL_EXT_abgr GL_EXT_bgra GL_EXT_bindable_uniform GL_EXT_blend_color GL_EXT_blend_equation_separate GL_EXT_blend_func_separate GL_EXT_blend_minmax GL_EXT_blend_subtract GL_EXT_compiled_vertex_array GL_EXT_Cg_shader GL_EXT_depth_bounds_test GL_EXT_direct_state_access GL_EXT_draw_buffers2 GL_EXT_draw_instanced GL_EXT_draw_range_elements GL_EXT_fog_coord GL_EXT_framebuffer_blit GL_EXT_framebuffer_multisample GL_EXTX_framebuffer_mixed_formats GL_EXT_framebuffer_multisample_blit_scaled GL_EXT_framebuffer_object GL_EXT_framebuffer_sRGB GL_EXT_geometry_shader4 GL_EXT_gpu_program_parameters 
[... cut ...]
#10 0x007f9d38315660 in (Unknown)
#11 0x007f9d38315670 in (Unknown)
#12 0x007f9d38315680 in (Unknown)
#13 0x007f9d38315690 in mono_print_method_from_ip
#14 0x007f9d38315c30 in mono_perfcounter_foreach
#15 0x007f9d38315c80 in mono_threads_detach_coop
#16 0x007f9d38315d60 in mono_unity_jit_cleanup
#17 0x007f9d38315dd0 in mono_threads_set_shutting_down
#18 0x007f9d38315ea0 in GC_inner_start_routine
#19 0x007f9d38315ef0 in GC_call_with_stack_base
#20 0x007f9d38315f20 in start_thread
#21 0x007f9d38315fc0 in clone
`

martins-mozeiko, 2019-05-09:
This seems to be Unity/Nvidia driver bug...
More information is here: https://github.com/Unity-Technologies/PostProcessing/issues/592

It is suggested to upgrade nvidia driver. Latest version for your GPU is v418, currently you have v390.

As alternative it suggests to try to run `export __GL_GlslUseCollapsedArrays=0` before launching binary. But I'm not sure what exactly does this affect...

fengsiyu, 2019-05-10:
I uninstall the old NVIDIA GPU Driver and install the verison 418.43 Driver. I have run the bootstrap and rosbridge in background. And when I run my scenarios setting script, the terminal which launched the simulator shows as usual:

`Stacktrace:
/proc/self/maps:
00400000-01d92000 r-xp 00000000 103:07 658033                            /home/fengsiyu/lgsvlsimulator/simulator
01f91000-02006000 r-xp 01991000 103:07 658033                            /home/fengsiyu/lgsvlsimulator/simulator
02006000-0203a000 rwxp 01a06000 103:07 658033                            /home/fengsiyu/lgsvlsimulator/simulator
0203a000-0211e000 rwxp 00000000 00:00 0 
02307000-3fefc000 rwxp 00000000 00:00 0                                  [heap]
4003a000-4003c000 r-xs 00000000 103:06 1703946                           /tmp/.glwvLig7 (deleted)
40078000-4021a000 rwxp 00000000 00:00 0 
40f2d000-40f3d000 rwxp 00000000 00:00 0 
411e0000-41292000 rwxp 00000000 00:00 0 
412c2000-41384000 rwxp 00000000 00:00 0 
7f7a28000000-7f7a2b40b000 rwxp 00000000 00:00 0 
7f7a2b40b000-7f7a2c000000 ---p 00000000 00:00 0 
7f7a30000000-7f7a33fe4000 rwxp 00000000 00:00 0 
7f7a33fe4000-7f7a34000000 ---p 00000000 00:00 0 
7f7a38000000-7f7a3b4be000 rwxp 00000000 00:00 0 
7f7a3b4be000-7f7a3c000000 ---p 00000000 00:00 0 
7f7a3c000000-7f7a3f4ed000 rwxp 00000000 00:00 0 
7f7a3f4ed000-7f7a40000000 ---p 00000000 00:00 0 
7f7a40000000-7f7a43777000 rwxp 00000000 00:00 0 
7f7a43777000-7f7a44000000 ---p 00000000 00:00 0 
7f7a44faa000-7f7a46f9d000 rwxp 00000000 00:00 0 
7f7a46f9d000-7f7a4739d000 rwxs 00000000 00:06 562                        /dev/nvidiactl
7f7a4739d000-7f7a48000000 rwxp 00000000 00:00 0 
7f7a48000000-7f7a4be81000 rwxp 00000000 00:00 0 
7f7a4be81000-7f7a4c000000 ---p 00000000 00:00 0 
7f7a4c000000-7f7a4fc4f000 rwxp 00000000 00:00 0 
7f7a4fc4f000-7f7a50000000 ---p 00000000 00:00 0 
7f7a50000000-7f7a5391f000 rwxp 00000000 00:00 0 
7f7a5391f000-7f7a54000000 ---p 00000000 00:00 0 
7f7a54009000-7f7a5b815000 rwxp 00000000 00:00 0 
7f7a5b878000-7f7a5f817000 rwxp 00000000 00:00 0 
7f7a5f817000-7f7a60000000 rwxs 00000000 00:06 562                        /dev/nvidiactl
7f7a60000000-7f7a63f4a000 rwxp 00000000 00:00 0 
7f7a63f4a000-7f7a64000000 ---p 00000000 00:00 0 
7f7a64000000-7f7a67fb7000 rwxp 00000000 00:00 0 
7f7a67fb7000-7f7a68000000 ---p 00000000 00:00 0 
7f7a68000000-7f7a6ba40000 rwxp 00000000 00:00 0 
7f7a6ba40000-7f7a6c000000 ---p 00000000 00:00 0 
7f7a6c000000-7f7a6ff77000 rwxp 00000000 00:00 0 
7f7a6ff77000-7f7a70000000 ---p 00000000 00:00 0 
7f7a70000000-7f7a73fdc000 rwxp 00000000 00:00 0 
7f7a73fdc000-7f7a74000000 ---p 00000000 00:00 0 
7f7a74000000-7f7a77ddf000 rwxp 00000000 00:00 0 
7f7a77ddf000-7f7a78000000 ---p 00000000 00:00 0 
7f7a78000000-7f7a7bdf1000 rwxp 00000000 00:00 0 
7f7a7bdf1000-7f7a7c000000 ---p 00000000 00:00 0 
7f7a7c000000-7f7a7fee7000 rwxp 00000000 00:00 0 
7f7a7fee7000-7f7a80000000 ---p 00000000 00:00 0 
7f7a80000000-7f7a83eeb000 rwxp 00000000 00:00 0 
7f7a83eeb000-7f7a84000000 ---p 00000000 00:00 0 
7f7a84031000-7f7a842fc000 rwxp 00000000 00:00 0 
7f7a842fc000-7f7a84ae5000 rwxs 00000000 00:06 562                        /dev/nvidiactl
7f7a84ae5000-7f7a852ef000 rwxp 00000000 00:00 0 
7f7a852ef000-7f7a85ad8000 rwxs 00000000 00:06 562                        /dev/nvidiactl
7f7a85ad8000-7f7a90000000 rwxp 00000000 00:00 0 
7f7a90000000-7f7a93e5f000 rwxp 00000000 00:00 0 
7f7a93e5f000-7f7a94000000 ---p 00000000 00:00 0 
7f7a94268000-7f7a95fff000 rwxp 00000000 00:00 0 
7f7a96051000-7f7a98000000 rwxp 00000000 00:00 0 
7f7a98000000-7f7a9be11000 rwxp 00000000 00:00 0 
7f7a9be11000-7f7a9c000000 ---p 00000000 00:00 0 
7f7a9c000000-7f7a9fe0e000 rwxp 00000000 00:00 0 
7f7a9fe0e000-7f7aa0000000 ---p 00000000 00:00 0 
7f7aa0000000-7f7aa3f5f000 rwxp 00000000 00:00 0 
7f7aa3f5f000-7f7aa4000000 ---p 00000000 00:00 0 
7f7aa4000000-7f7aa7dec000 rwxp 00000000 00:00 0 
[... 1443 lines cut ...]
#6  0x00007f7be92adbb5 in mono_exception_from_name_domain () from /home/fengsiyu/lgsvlsimulator/simulator_Data/MonoBleedingEdge/x86_64/libmonobdwgc-2.0.so
#7  0x00007f7be91ff4bd in ?? () from /home/fengsiyu/lgsvlsimulator/simulator_Data/MonoBleedingEdge/x86_64/libmonobdwgc-2.0.so
#8  0x00007f7be9273b3a in ?? () from /home/fengsiyu/lgsvlsimulator/simulator_Data/MonoBleedingEdge/x86_64/libmonobdwgc-2.0.so
#9  0x00000000401c4293 in ?? ()
#10 0x0000000000000000 in ?? ()

=================================================================
Got a SIGABRT while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================

Aborted
`
And the rosbridge terminal shows as below:
`/use_sim_time is not set, will not subscribe to simulated time [/clock] topic
/apollo/ros_pkgs/src/rosbridge_suite/rosbridge_server/scripts/rosbridge_websocket:218 <module>() [INFO] [1557487531.350063]: Rosbridge WebSocket server started on port 9090
/apollo/ros_pkgs/src/rosbridge_suite/rosbridge_server/src/rosbridge_server/websocket_handler.py:77 open() [INFO] [1557487555.568935]: Client connected.  1 clients total.
the rosdep view is empty: call 'sudo rosdep init' and 'rosdep update'
/apollo/ros_pkgs/src/rosbridge_suite/rosbridge_library/src/rosbridge_library/protocol.py:385 log() [INFO] [1557487556.326352]: [Client 0] Subscribed to /apollo/control
[... cut ...] 56.341816]: [Client 0] Subscribed to /simulator/reset
/apollo/ros_pkgs/src/rosbridge_suite/rosbridge_server/src/rosbridge_server/websocket_handler.py:112 on_close() [INFO] [1557487557.730301]: Client disconnected. 0 clients total.
`
What should I do? I'm quiet worried about my experiment and I want to solve this problem as soon as possible. Could you help me solving this problem? 
Have you ever using the python script to set up scenarios and then using the Apollo platform to run the simulation ?

fengsiyu, 2019-05-10:
I want to konw when I update the NVIDIA Graphic Driver, should I update the CUDA to version 10? (At present time, my CUDA version is: 9.0.176)

martins-mozeiko, 2019-05-10:
Have you tried running simulator with `export __GL_GlslUseCollapsedArrays=0` ?

CUDA SDK version does not matter. Simulator or Apollo are not using cuda SDK. You can install 10 or leave it 9, it won't change a thing.

martins-mozeiko, 2019-05-10:
> Have you ever using the python script to set up scenarios and then using the Apollo platform to run the simulation ?

We have released more examples for API that shows how to set up specific scenarios for running with Apollo: https://www.lgsvlsimulator.com/docs/api-example-descriptions/

fengsiyu, 2019-05-11:
I have tried to run simulator with export __GL_GlslUseCollapsedArrays=0, but the same error occurs as usual. What should I do? 
Here is my python script:
`#!/usr/bin/env python3
# name:直行双车道上的场景设置
# author: Fengsiyu
# version: v1.0

# 导入相关的模块包
import os
import lgsvl
import random
import math
import time

# 设定模拟器的IP和端口号
sim = lgsvl.Simulator(os.environ.get("SIMULATOR_HOST", "192.168.1.104"), 8181)

# 若当前模拟器场景为旧金山，则模拟器重启，否则加载旧金山场景
if sim.current_scene == "SanFrancisco":
  sim.reset()
else:
  # 使用load方法加载场景地图：旧金山
  sim.load("SanFrancisco")

# 获取车辆初始位置数据
spawns = sim.get_spawn()

# 获取车辆状态对象
state = lgsvl.AgentState()

# spawns[0]：自车位于左边车道位置
# spawns[1]：自车位于右边车道位置
# Unity坐标系，左手坐标系，y是up, z是north, x是east
# 选择最长的直行道作为本车的初始位置(223.809997558594, 10.1000003814697, 4.3500061035159945)
ego_sx = spawns[0].position.x + 13
ego_sy = spawns[0].position.y
ego_sz = spawns[0].position.z - 193.5

# rotation:(0.0159243624657393, 269.949066162109, 3.56300406565424e-05)
ego_rx = spawns[0].rotation.x
ego_ry = spawns[0].rotation.y
ego_rz = spawns[0].rotation.z

# 本车初始位置
state.transform.position.x = ego_sx
state.transform.position.y = ego_sy
state.transform.position.z = ego_sz

# 本车初始旋转矢量
state.transform.rotation.x = ego_rx
state.transform.rotation.y = ego_ry
state.transform.rotation.z = ego_rz

# 添加本车智能体
a = sim.add_agent("XE_Rigged-apollo", lgsvl.AgentType.EGO, state)

# 添加NPC车辆(场景1)
# 本车前方20米有障碍车
obs_veh1_sx = ego_sx - 20.0
obs_veh1_sy = ego_sy
obs_veh1_sz = ego_sz
state.transform.position.x = obs_veh1_sx
obs_veh1 = sim.add_agent("Sedan", lgsvl.AgentType.NPC, state)

# 车前障碍车按照预先路点运行
waypoints = []
obs_veh1_speed = 5.55
for i in range(62):
    px = (i+1) * 5
    wp = lgsvl.DriveWaypoint(lgsvl.Vector(obs_veh1_sx - px, obs_veh1_sy, obs_veh1_sz),obs_veh1_speed)
    waypoints.append(wp)

def on_waypoint(agent, index):
    print("waypoint {} reached".format(index))
obs_veh1.on_waypoint_reached(on_waypoint)
obs_veh1.follow(waypoints)

# 本车前方120米有障碍车
state.transform.position.x = ego_sx - 300.0
obs_veh2 = sim.add_agent("Sedan", lgsvl.AgentType.NPC, state)

state.transform.position.x = ego_sx - 300.0
state.transform.position.z = ego_sz + 3.5
obs_veh3 = sim.add_agent("Sedan", lgsvl.AgentType.NPC, state)

# 左边双车道添加NPC车辆
state.transform.position.x = ego_sx - 200.0
state.transform.position.z = ego_sz - 3.5
obs_veh4 = sim.add_agent("Sedan", lgsvl.AgentType.NPC, state)

state.transform.position.x = ego_sx - 220.0
state.transform.position.z = ego_sz - 7
obs_veh5 = sim.add_agent("Sedan", lgsvl.AgentType.NPC, state)

# 设置障碍物车辆跟随车道的最大速度
# 单位：米/秒 5.55m/s = 20km/h
#obs_veh1.follow_closest_lane(True, 5.55)
obs_veh2.follow_closest_lane(True, 5.55)
obs_veh3.follow_closest_lane(True, 5.55)
obs_veh4.follow_closest_lane(True, 5.55)
obs_veh5.follow_closest_lane(True, 5.55)

# 获取传感器状态
sensors = a.get_sensors()

# 打开摄像头传感器
for s in sensors:
  if isinstance(s, lgsvl.CameraSensor):
    s.enabled = True

# 打开激光雷达传感器
for s in sensors:
  if isinstance(s, lgsvl.LidarSensor):
    s.enabled = True

# 打开IMU传感器
for s in sensors:
    if isinstance(s, lgsvl.ImuSensor):
        s.enabled = True
# 连接到bridge
a.connect_bridge("192.168.1.104", 9090)
print("Waiting for connection...")

while not a.bridge_connected:
    # 休眠1秒钟
    time.sleep(1)

print("Bridge connected:", a.bridge_connected)

print("wait several seconds(30s) to wait Dreamview setting destination point...")
time.sleep(30)

# 运行仿真
print("run the simulation...")
sim.run()
`
Could you help me to read my python script and find wether it has some errors?
Thank you for your kindly help.

martins-mozeiko, 2019-05-13:
I was able to reproduce problem. Currently I'm not sure what is wrong.
You may want to try to enable only Main and Telephoto cameras instead of enabling all cameras:
```
for s in sensors:
  if isinstance(s, lgsvl.CameraSensor):
    if s.name == "Main Camera" or s.name == "Telephoto Camera":
      s.enabled = True
```
Maybe this will help...

We'll try to fix this for next release of simulator.

fengsiyu, 2019-05-14:
Thank you for helping me to solve this problem. When I only enable the Main and Telephoto cameras, the simulator doesn't crash.

I am looking forward to your new release of simulator.

outmanlol, 2019-06-11:
@martins-mozeiko,@fengsiyu :
    hello,I have a question. I tied to change the speed of the npc vehicle with the scenario-npc-lane-change.py and the python file above,Neither works。can you give me some advices?
 Thank you for your answer.

martins-mozeiko, 2019-06-11:
@outmanlol what version of simulator are you using. This is supposed to be fixed in 2019.05 release. Have you tried it scripts with latest version of simulator?

outmanlol, 2019-06-12:
> @outmanlol what version of simulator are you using. This is supposed to be fixed in 2019.05 release. Have you tried it scripts with latest version of simulator?

Thinks for your reply. when I use the version which released in 2019.05,it is ok.

closed 2019-06-13 by martins-mozeiko