lgsvl/simulator #150: Havenot some item in MapToolUtilEdit

pengpingliang, 2019-05-08:
I try to install the unity(2018.2.4f1) and make a map with  the guide「https://github.com/lgsvl/simulator/blob/master/Docs/getting-started.md」
and
「https://github.com/lgsvl/simulator/blob/master/Docs/map-annotation.md」
, but the following 4 items arenot  shown in my Tool。
1.「Nullify All Nerihbor Lane Fields」
2.「Link Neighbor Lanes from Left」
3.「Link Neighbor Lanes from Right」
4.「Link Reverse Neighbor Lanes」

Q: Have any wrong ? and , How can i resolve it ?

EricBoiseLGSVL, 2019-05-09:
@pengpingliang 

Hello.  These fields are now calculated automatically when you export or on start for NPCs.  No need to fill them in anymore and the buttons are removed.  We will be updating our documentation soon.

-Eric

pengpingliang, 2019-05-10:
@EricBoiseLGSVL 
I got it, and thank you very much!

closed 2019-05-14 by pengpingliang