lgsvl/simulator #151: Is it possible to release the Python APIs as a submodule?

haisongdong-harrison, 2019-05-09:
The new Python APIs look really nice. But due to the fact that it's now a part of the simulator repository, anyone who wants to use it must clone the whole repo even though they may not need the source code of the simulator.

Is it possible to make the APIs a separate submodule? It would be a lot easier for the users to integrate into their own work without too much overhead added to the simulator project (only one additional `git submodule update` needed after clone/pull).

martins-mozeiko, 2019-05-09:
This is a good suggestion. We'll update repository to have api as submodule.

martins-mozeiko, 2019-08-19:
PythonAPI now is moved to separate repository: https://github.com/lgsvl/PythonAPI

closed 2019-08-19 by martins-mozeiko