lgsvl/simulator #152: Cannot install Unity on Ubuntu 16.04

bigsheep2018, 2019-05-09:
Hello lgsvl team,
Thanks for sharing this work.
I can download the released files and run it without any problem. But I got some issues when I am trying to build the source on Ubuntu.

I followed the instructions [here](https://www.lgsvlsimulator.com/docs/build-instructions/#instructions-to-build-standalone-executable)

Download UnitySetup-2018.2.4f1 [here](https://beta.unity3d.com/download/fe703c5165de/public_download.html)
Give permissions: chmod +x ~/Downloads/UnitySetup-2018.2.4f1
Run it with:
Sudo ./UnitySetup-2018.2.4f1 --unattended --install-location=/opt/Unity --components=Unity,Windows-Mono

I got the following errors:
`Do you accept the terms of the License Agreement? (y/n)
y

Beginning unattended installation to '/opt/Unity', downloading packages to '<temporary location>'
Error 404 getting 'https://netstorage.unity3d.com/unity/fe703c5165de/unity-2018.2.4f1-linux.ini'
Download of 'https://netstorage.unity3d.com/unity/fe703c5165de/unity-2018.2.4f1-linux.ini' failed
Successfully downloaded https://download.unity3d.com/download_unity/fe703c5165de/unity-2018.2.4f1-linux.ini => /tmp/.SX0L1Z
Selecting Unity
Selecting Windows-Mono
Required installation size: Total space required: 3.93 GB
Available space: Space available: 85.04 GB
Available space: Space available: 85.04 GB
Downloading Unity 2018.2.4f1
Error 424 getting 'https://download.unity3d.com/download_unity/fe703c5165de/LinuxEditorInstaller/Unity.tar.xz'
Download of 'https://download.unity3d.com/download_unity/fe703c5165de/LinuxEditorInstaller/Unity.tar.xz' failed
Error downloading 'https://download.unity3d.com/download_unity/fe703c5165de/LinuxEditorInstaller/Unity.tar.xz' to '/tmp/Unity.tar.xz'`

Any suggestions for solving this problem?
Thanks in advance.

Mushekwi, 2019-05-09:
I had the same problem before,just change to a higher version.
Please refer to https://forum.unity.com/threads/unity-on-linux-release-notes-and-known-issues.350256/page-2
But I'm not sure if there are any other problems.

bigsheep2018, 2019-05-09:
Hello @Mushekwi,
Thanks for the suggestion. 
As you suggested, I tried UnitySetup-2018.2.5f1, UnitySetup-2018.2.6f1, UnitySetup-2018.2.7f1, and, it seems UnitySetup-2018.2.6f1 works.

Then the following error occurs:
Do you accept the terms of the License Agreement? (y/n)
y

Beginning unattended installation to '/opt/Unity', downloading packages to '<temporary location>'
Error 404 getting 'https://netstorage.unity3d.com/unity/793261fe3e9a/unity-2018.2.6f1-linux.ini'
Download of 'https://netstorage.unity3d.com/unity/793261fe3e9a/unity-2018.2.6f1-linux.ini' failed
Error 424 getting 'https://download.unity3d.com/download_unity/793261fe3e9a/unity-2018.2.6f1-linux.ini'
Download of 'https://download.unity3d.com/download_unity/793261fe3e9a/unity-2018.2.6f1-linux.ini' failed
Successfully downloaded https://beta.unity3d.com/download/793261fe3e9a/unity-2018.2.6f1-linux.ini => /tmp/.5XFK1Z
Selecting Unity
Selecting Windows-Mono
Required installation size: Total space required: 3.93 GB
Available space: Space available: 82.17 GB
Available space: Space available: 82.17 GB
Downloading Unity 2018.2.6f1
Finished downloading 'https://beta.unity3d.com/download/793261fe3e9a/LinuxEditorInstaller/Unity.tar.xz' to '/tmp/Unity.tar.xz'
Downloading Windows Build Support (Mono)
Checksum for downloaded '/tmp/UnitySetup-Windows-Mono-Support-for-Editor-2018.2.6f1.pkg' (eb13328e6429b5c84b4489a0221d5011) doesn't match expectation (9476294e877ef2def4b29b9bbf76f3ed)Finished downloading 'https://beta.unity3d.com/download/793261fe3e9a/MacEditorTargetInstaller/UnitySetup-Windows-Mono-Support-for-Editor-2018.2.6f1.pkg' to '/tmp/UnitySetup-Windows-Mono-Support-for-Editor-2018.2.6f1.pkg'
Installing Unity 2018.2.6f1
Preparing Windows Build Support (Mono)
Error opening archive /tmp/UnitySetup-Windows-Mono-Support-for-Editor-2018.2.6f1.pkg: Failed to open '/tmp/UnitySetup-Windows-Mono-Support-for-Editor-2018.2.6f1.pkg'
Error: Missing payload in package
Installation failed, aborting``

I found a [link](https://forum.unity.com/threads/installation-error-what-should-i-do.526184/) here mentioned this error, it says missing dependencies for libarchive.
I have installed all the mentioned package mentioned in the link you provided including libarchive, and I still get this error.
Any further suggestions?

Thanks.

bigsheep2018, 2019-05-09:
@martins-mozeiko  
Update---------------------
I canceled installation of Windows-Mono component, which means using the following command:
Sudo ./UnitySetup-2018.2.4f1 --unattended --install-location=/opt/Unity --components=Unity

I then can install Unity successfully. May I know the reason? or would there be any further problem?or any methods that fix this problem?
Thanks.

martins-mozeiko, 2019-05-09:
Windows-Mono is needed only if you want to build standalone executable for Windows. If all you will do is run Simulator on Ubuntu, then you don't need Windows-Mono. Everything else will work fine.

Error 424 sounds like some kind of issue with http download. Are you using some kind of proxy/vpn/firewall? Maybe your network or ISP blocks the download?

bigsheep2018, 2019-05-10:
Thanks for the reply. @martins-mozeiko 
I install UnitySetup-2018.3.0f2 and all above mentioned problems are fixed (including downloading and windows-mono issues). I have not test all the functionality of the simulator, but I can run the lgsvl-car in San Francisco city at least.

I googled about Unity3d-2018.2.* a lot recently : ( and found there might be more issues such as payload missing, [compile errors](https://forum.unity.com/threads/problem-when-updating-and-changing-the-installation-location.554881/) , etc. I think Unity3d-2018.2.* might not be a stable version on Ubuntu.

Considering there might be a branch of people in similar situation suffering from installing Unity3d on Ubuntu, I suggest upgrade the unity3D to 2018.3.* or higher versions.

martins-mozeiko, 2019-05-10:
Pretty much all versions of Unity have one or another issue. There is no good one version that works without issues.

Migrating to 2018.3 or even 2019.1 is in our plans, but that will not happen fast. It requires verifying that everything else works and nothing is broken. For example 2018.3 completely changed how prefabs work, so upgrade to it is a large change.

In general it is possible to download unity installer offline, and once you have, you should be able to install it on any machine.

bigsheep2018, 2019-05-17:
Thanks. Close the issue

closed 2019-05-17 by bigsheep2018

outmanlol, 2019-05-27:
hello  @bigsheep2018 ,@martins-mozeiko 
I had the same problem,when i  try to build UnitySetup-2018.2.4f1 with the following command:
sudo ./UnitySetup-2018.2.4f1 --unattended --install-location=/opt/Unity --components=Unity

I got the same error(I have tried in many different net )：
Error 404 getting 'https://netstorage.unity3d.com/unity/fe703c5165de/unity-2018.2.4f1-linux.ini'

So I try to build UnitySetup-2018.3.0f2 ,it succeed.
and run the Unity in the path /opt/Unity/Editor ,make sure it is activated.
next,I change into the file path of simulator,run build:
 mkdir build
 /opt/Unity/Editor/Unity \
    -batchmode \
    -nographics \
    -silent-crashes \
    -quit \
    -buildDestination ./build/simulator \
    -buildTarget Linux64 \
    -executeMethod BuildScript.Build \
    -projectPath . \
    -logFile /dev/stdout

I got the error:
Scripts have compiler errors.
(Filename: /home/builduser/buildslave/unity/build/Runtime/Utilities/Argv.cpp Line: 348)
Aborting batchmode due to failure:
Scripts have compiler errors.

Any suggestions for solving this problem?
Thanks in advance.

martins-mozeiko, 2019-05-27:
@outmanlol 
> Error 404 getting 'https://netstorage.unity3d.com/unity/fe703c5165de/unity-2018.2.4f1-linux.ini'

This message is normal and expected. It is no an error when installing Unity. If installation fails then real error is shown later, somewhere below this message.

To use different major version of Unity (like 2018.3, or 2018.4 or 2019.1) you will need to open project in Unity and fix compilation errors. Unity changes API form version to version and you need to fix incompatible changes manually.

You really should use 2018.2 version. Preferably 2018.2.4. Using other version might require manual fixes to code and/or scene/prefabs.

outmanlol, 2019-05-27:
[@martins-mozeiko](url) Thanks for the reply.
when i ignal the Error 404 ,it succeed.