lgsvl/simulator #153: Are there RVIZ plugins for custom msgs like lgsvl_msgs?

opoppo, 2019-05-10:
Just found out this simulator is really amazing work!
A small question, is there an RVIZ plugin for custom msgs like lgsvl_msgs ground truth boxes, or they can only be visualized within the simulator?

martins-mozeiko, 2019-05-10:
We don't have rviz plugin for this. Currently we have used only rostopic echo to "see" this data.

But this sounds something that would very useful. If you are willing to implement it, we'll gladly accept pull request :)

opoppo, 2019-05-11:
I would be glad to contribute to the project, but it won't be before my thesis program is settled.   
Still, thanks for the reply! Rostopic echo is good enough for me so far.

closed 2019-05-11 by opoppo

opoppo, 2019-05-18:
Hey, I wrote this ROS node for a quick relay to jsk_recognition_msgs/BoundingBoxArray. It might be helpful if anyone needs to visualize the boxes in RVIZ for a quick look like me. 

`#!/usr/bin/env python

import rospy
from jsk_recognition_msgs.msg import BoundingBox, BoundingBoxArray
from lgsvl_msgs.msg import Detection3D, Detection3DArray, BoundingBox3D
import std_msgs.msg

relayed_box_array = BoundingBoxArray()

def talker():
    global relayed_box_array
    pub = rospy.Publisher('relayed_bbox', BoundingBoxArray, queue_size=10)
    # rospy.init_node('talker', anonymous=True)
    # rate = rospy.Rate(5)  # 10hz
    # while not rospy.is_shutdown():
    num=len(relayed_box_array.boxes)
    if num>0:
        hello_str = "showing bbox %s" % num
        rospy.loginfo(hello_str)
        relayed_box_array.header.stamp=rospy.get_rostime()
        relayed_box_array.header.frame_id='velodyne'
        pub.publish(relayed_box_array)

def relay(detection_array):
    # rate = rospy.Rate(20)
    global relayed_box_array
    relayed_box_array.header.frame_id = 'velodyne'
    relayed_box_array.header.stamp = detection_array.header.stamp
    for det in detection_array.detections:
        relayed_box = BoundingBox()
        relayed_box.header.frame_id = 'velodyne'
        relayed_box.pose = det.bbox.position
        relayed_box.dimensions = det.bbox.size
        relayed_box.header = det.header
        relayed_box_array.boxes.append(relayed_box)
        # pub.publish(relayed_box)

    # pub.publish(relayed_box_array)
    # print('GT target num ', len(relayed_box_array.boxes))
    # rate.sleep()
    talker()
    relayed_box_array.boxes=[]

def listener():
    rospy.init_node('bbox_relay', anonymous=True)

    rospy.Subscriber('simulator/ground_truth/3d_detections', Detection3DArray, relay)

    rospy.spin()

if __name__ == '__main__':
    try:
        listener()
        # talker()
    except rospy.ROSInterruptException:
        pass
`

[image]

opoppo, 2019-05-18:
Hey, I wrote this ROS node for a quick relay to jsk_recognition_msgs/BoundingBoxArray. It might be helpful if anyone needs to visualize the boxes in RVIZ for a quick look like me.
[image]
`
#!/usr/bin/env python

import rospy
from jsk_recognition_msgs.msg import BoundingBox, BoundingBoxArray
from lgsvl_msgs.msg import Detection3D, Detection3DArray, BoundingBox3D
import std_msgs.msg

relayed_box_array = BoundingBoxArray()

def talker():
    global relayed_box_array
    pub = rospy.Publisher('relayed_bbox', BoundingBoxArray, queue_size=10)
    num=len(relayed_box_array.boxes)
    if num>0:
        hello_str = "showing bbox %s" % num
        rospy.loginfo(hello_str)
        relayed_box_array.header.stamp=rospy.get_rostime()
        relayed_box_array.header.frame_id='velodyne'
        pub.publish(relayed_box_array)

def relay(detection_array):
    global relayed_box_array
    relayed_box_array.header.frame_id = 'velodyne'
    relayed_box_array.header.stamp = detection_array.header.stamp
    for det in detection_array.detections:
        relayed_box = BoundingBox()
        relayed_box.header.frame_id = 'velodyne'
        relayed_box.pose = det.bbox.position
        relayed_box.dimensions = det.bbox.size
        relayed_box.header = det.header
        relayed_box_array.boxes.append(relayed_box)
    talker()
    relayed_box_array.boxes=[]

def listener():
    rospy.init_node('bbox_relay', anonymous=True)
    rospy.Subscriber('simulator/ground_truth/3d_detections', Detection3DArray, relay)
    rospy.spin()

if __name__ == '__main__':
    try:
        listener()
    except rospy.ROSInterruptException:
        pass
`

ErijBNT, 2019-05-22:
@opoppo 
I am using ros1 
I want to know how did you dealt with this error while connecting to rosbridge.

[image]

I also want to subscribe to the camera node in order to visualize it in Rviz and to make object detection in order to drive the car autonomously.

I really need help.

opoppo, 2019-05-23:
> @opoppo
> I am using ros1
> I want to know how did you dealt with this error while connecting to rosbridge.
> 
> [image]
> 
> I also want to subscribe to the camera node in order to visualize it in Rviz and to make object detection in order to drive the car autonomously.
> 
> I really need help.

Did you install the pb-msgs package?
Actually, I was using the autoware model, so i installed nmea-msgs, autoware-msgs and lgsvl msgs to get rosbridge work. In your case, it seems to be pb-msgs from apollo git repo. Hope that works?

opoppo, 2019-05-23:
@ErijBNT 
#93 
Check this out, guess it may help?

martins-mozeiko, 2019-05-23:
This error is harmless if you are not using Apollo software. rosbridge will still work fine for all other messages. `pb_msgs` are needed only if you connect Simulator to Apollo environment. To get pb_msgs you need to run rosbridge inside Apollo 3.0 ROS environment.

ErijBNT, 2019-05-23:
@martins-mozeiko thanks
@opoppo And concerning the Rviz part. how did you manage to vizualize the environment on Rviz?

opoppo, 2019-05-24:
@ErijBNT 
I subscribed to the ros topics and there they are.

ErijBNT, 2019-05-24:
@opoppo is it possible to show me a script that subscribes to the main camera in order to view it on rviz
I didn't manage to do that

ErijBNT, 2019-05-24:
@opoppo is it possible to generate an occupancy grid map from the lidar?

crankler, 2020-08-27:
@opoppo  HI, thank you for your share, 
when I use this code, and add  jsk_recognition_msgs/BoundingBoxArray, I get the error of rviz:
`[ERROR] [1598542990.471080399]: Error transforming pose from frame '' to frame 'velodyne'
`
and I find the topic as follow;

```
header: 
  seq: 326
  stamp: 
    secs: 1598543032
    nsecs:  43184995
  frame_id: "base_link"
boxes: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: ''
    pose: 
      position: 
        x: -6.31709098816
        y: 10.7571353912
        z: -1.97511124611
      orientation: 
        x: -2.8196784987e-08
        y: 3.82501219711e-07
        z: -0.999994834437
        w: 0.00321420273532
    dimensions: 
      x: 0.5
      y: 0.5
      z: 2.0
    value: 0.0
    label: 0
```