lgsvl/simulator #154: Error while Save/restore vehicle position.

kkhitsko, 2019-05-10:
There is some error while I try to reset to last saved position of vehicle by pressing F9 button. Restored position is different from last saved position.

martins-mozeiko, 2019-05-10:
By "saved position" you mean position when you press F5?
How different is the restored position? Is it nearby where you saved, somewhere else completely?

kkhitsko, 2019-05-14:
> By "saved position" you mean position when you press F5?

Yes

> How different is the restored position? Is it nearby where you saved, somewhere else completely?

x and y coordinates of restored position almost equal to the saved position, by z is mush less than zero and constantly decreasing. It's looks like that car falls.

[image]

But If I press "Vehicle position reset" button, vehicle position is restored correctly.

closed 2020-04-07 by EricBoiseLGSVL