lgsvl/simulator #155: Logitech G29 steering wheel Support

aravindSwamy94, 2019-05-13:
Hello team, 
                        How to integrate logitech g29 steering wheel with LG simulator, already there is an interface available for G920 . will there be similar SDL_GAMECONTROLLERCONFIG for G29 steering wheel also??? 

Please help me on this issue

Regards,
Aravind S

martins-mozeiko, 2019-05-13:
Any wheel should work as long as SDL_GAMECONTROLLERCONFIG setting is set up correctly.

You can create this setting with tools like [SDL2 ControllerMap](https://www.libsdl.org/download-2.0.php) (from libSDL) or [SDL2 Gamepad Tool](http://generalarcade.com/gamepadtool/) or [Steam](https://github.com/gabomdq/SDL_GameControllerDB#steam).

aravindSwamy94, 2019-05-14:
Hi @martins-mozeiko : Thanks for your reply. 

                            I tried creating SDL_GAMECONTROLLERCONFIG for logitech G29 and im unable to do that. Using the tools mentioned above, I tried to install SDL2 and SDL2 Gamepad tool onto my system , but the run- time version of the code crashes. Im getting a segementation fault , if I run SDL2 Gamepad tool . 
can you please help me on this

System Used : Ubuntu 16.04 , 64-bit system

michaelhany55, 2019-11-22:
I'm also running on Ubuntu (18.04 though) and I have a logitech G27 wheel. I created the SDL_GAMECONTROLLERCONFIG environment variable. However; the simulator still wouldn't take commands from the wheel. I can use my keyboard to move the vehicle but not the wheel. Is there something missing? Is there like a guide I can follow to set it up? lastly, is there a sensor I should add to the vehicle to allow it to take commands from my Logitech G27 wheel?
Thanks in advance for your help.

martins-mozeiko, 2019-11-22:
Are you using old simulator, or new HDRP one?

For new simulator only G920 steering wheel is supported. To support other steering wheels you'll need to modify Unity Input Systems configuration and rebuild simulator binary.  We are currently creating plugin system that will allow to add this kind of functionality that will not require rebuilding simulator.

AchimTuran, 2019-12-14:
@martins-mozeiko 
Can you please provide further information what needs to be done? I'm currenlty having the same issue but I don't know exactly what needs to be changed to get the Logitech G29 running with the simulator. I'm currenlty using 2019.11.

I currently don't know what needs to be mapped because the names in my Unity project are not very clear to me.
[image]

AchimTuran, 2019-12-14:
Okay found it. :)

You need to do a double click on Assets/Scripts/Controllers/SimulatorControls.inputactions
[image]

Then you can change the buttons. Currently trying if it will work on my machine.
[image]

AchimTuran, 2019-12-14:
Okay it worked, I'm now able to drive with my Logitech G29 wheel. 😃 

These are the changes I did:
[Logitech_G29_support.zip](https://github.com/lgsvl/simulator/files/3964255/Logitech_G29_support.zip)
[image]
[image]

whydaydayup, 2019-12-28:
Hi @martins-mozeiko
Have you created plugin system that allow to add this kind of functionality that will not require rebuilding simulator.Now I have rebuild the simulator, but I can only rebuild the open source map and vehicle. The other map and vehicle provided by your website is not working well.

Best Regards!

martins-mozeiko, 2019-12-28:
We have not implemented this as plugin yet.

But rebuilding just simulator binary should work fine with our current assets - maps and vehicles. Just make sure you use source code from git tag for same version as for assets. We push weekly updates to git master that may introduce incompatibilities to assets, so don't use that if you cannot rebuild assets.

closed 2020-04-07 by EricBoiseLGSVL

ArteryLee, 2020-10-18:
> Okay it worked, I'm now able to drive with my Logitech G29 wheel. 😃
> 
> These are the changes I did:
> [Logitech_G29_support.zip](https://github.com/lgsvl/simulator/files/3964255/Logitech_G29_support.zip)
> [image]
> [image]

Hi! Our team is now facing the same issue and we have tried your method. However, it does not work in our system. May I ask if you can provide the whole Build?

EricBoiseLGSVL, 2020-10-19:
@ArteryLee 
- Be sure to have the wheel control sensor attached to the vehicle in the sensor parameters. 
- Are you able to open in Editor Window -> Analysis -> Input Debugger?  This will show you that the wheel is being recognized by Unity.  If it isn't then you have issues installing the Wheel drivers or software and need to resolve this before the wheel inputs will work.
- Finally, the file shared is all you need, just replace the file in simulator with it and you will have the same input settings or manually change them.

Manoj-EK, 2021-08-31:
> Are you using old simulator, or new HDRP one?
>
> For new simulator only G920 steering wheel is supported. To support other steering wheels you'll need to modify Unity Input Systems configuration and [...]

@martins-mozeiko is the plug-in system created?

EricBoiseLGSVL, 2021-08-31:
We are currently working on this.  We hope to release soon