lgsvl/simulator #156: Feature request - Different types of traffic lights

krishnatoshniwal, 2019-05-13:
Can different types of traffic lights be added on the map? I wanted to add 4 and 5 channel traffic lights to make some sophisticated intersections and test the stack. How can I go about doing this, is this in the development pipeline by any chance?

martins-mozeiko, 2019-05-13:
Currently we do not support this. Adding this feature would require to create 3D mesh for desired look, then changing code that manages signal color, and updating HDMap export functionality so Apollo or Autoware can get information about new traffic lights. Not sure if they will be to use correctly though.

krishnatoshniwal, 2019-05-13:
Got it, thank you!

EricBoiseLGSVL, 2020-04-07:
@krishnatoshniwal we are currently working on this and hope to include in a release soon

closed 2020-04-07 by EricBoiseLGSVL