lgsvl/simulator #157: collect data from sensors and ros1

ErijBNT, 2019-05-13:
I want to build my own project using ros melodic.

I want to collect data from the sensors(Camera, radar and lidar) of the car XE_Rigged in order to process this data to generate an occupancy grid map and visualized using Rviz. What are the steps to do that?
Well, I connected the simulator with the rosbridge. So, these errors appears.
[image]

I run rostopic list, this is the output.
[image]

When I run the rviz 
I cannot vizualize data

luqiang21, 2019-05-13:
@ErijBNT Are you using the correct vehicle in simulator?

ErijBNT, 2019-05-13:
What do you mean by the right vehicle? I select the xe_rigged in the simulator when I run the simulator

martins-mozeiko, 2019-05-13:
@ErijBNT The vehicles we currently have are preconfigured for publishing messages in AD specific stack formats. This includes choice of protocol - ROS1, ROS2, CyberRT. And message types and topic names.

For example, if you chouuse "XE_Rigged-apollo" vehicle, the sensor code will publish ROS & Protobuf messages to ROS with Apollo environment. If you'll choose "XE_Rigged-autoware" vehicle, then the sensors will publish Autoware specific messages to ROS1 environment.

If you want to run simulator without Autoware or Apollo environments then you will need to modify sensor code to publish messages in format you want. You can check any existing sensor code to see how it decides in what format it needs to publish message and then add extra condition to format message data however you need it.

Other option is to install Autoware message types in your ROS environment - then you'll be able to use simulator without modifcations with "XE_Rigged-autoware" vehicle. Only thing you'll need to do is to subscribe to Autoware topics and use Autoware message types to receive sensor information and publish back control.

ErijBNT, 2019-05-13:
@martins-mozeiko  I want to run simulator without Autoware or Apollo environments. Is there any tutorial for how to modify the sensor code ?? I am a beginner and I need help

martins-mozeiko, 2019-05-15:
I suggest to look into our [lanefollowing](https://github.com/lgsvl/lanefollowing) project which shows how to control car with DNN trained from camera image. ROS AD stack there receives only image and sends back control command. It does not use Apollo or Autoware at all.

ErijBNT, 2019-05-15:
@martins-mozeiko   I did all the steps with success in the lanefollowing project(data collect, preprocess and train). when I run "docker-compose up drive" and enable the main camera, it works correcly  in the terminal as shown in the tuto, but, the vehicle in the simulator doesn't move. what's the problem?
Added to that, I want to know how is it possible to add code to this project, so the vehicle will recieve data from many sensors and see the visualization in Rviz?

martins-mozeiko, 2019-05-16:
Does simulator says that bridge is "Connected" ?
Can you check with `ros2 topic echo /simulator/control/command" to check if ROS2 is sending control commands to drive?

To add new code for publishing new data messages to send over ROS you need to check how existing sensors are done and write similar that will publish what you need. There is really no guide for this, because it heavily depends on what kind of data and how to collect it.. You should read existing sensor code to understand what is needed.

closed 2020-04-07 by EricBoiseLGSVL