lgsvl/simulator #158: issues  using duckietown_msgs

kissofwind, 2019-05-14:
hey, lg team,

 I tried to run a ROS demo with duckietown_msgs.  what I was doing :   ubuntu 16.04 
      
1)   built(source ROS2 + python3.7) the duckietown_msgs (git clone from duckietown2) in a seperate folder (*.idl) and add the folder path to $PYTHONPATH 

2)   started the simulator in terminal A , map=  duckietown;  the car= duckietown-ros1,
and " roslaunch rosbridge_server rosbridge_websocket.launch"   in terminal B 

there are two concerns: 

1)  looks `roslaunch rosbridge_server` is only in ROS1 env(lunar),   `duciketown_msgs` is built by `ament` tool, which depends on python3.5+,  finally make the `duckietown_msgs`  not comptiable when running  rosbridge_websocket.  

    I wonder if rebuild simulator from source and configure python3.5, ROS2 env, will it be fixed ?

2)  there is another error:
[Client 0] subscribe: Unable to load the manifest for package duckietown_msgs. Caused by: duckietown_msgs

I suppose the server don't understand `duckietown_msgs`,  for this kind of external message type, how to register in server ?

there is [build_ros2_lgsvl](https://github.com/lgsvl/build-ros2-lgsvl), doesn't get it. 

thanks for explaining

martins-mozeiko, 2019-05-15:
1. To run ros bridge for ROS2 you need to use this project: https://github.com/RobotWebTools/ros2-web-bridge As you already are saying rosbridge_server package is ROS1 only.

2. duckietown_msgs should be available when you build packages from https://github.com/lgsvl/duckietown2

kissofwind, 2019-05-20:
>     1. To run ros bridge for ROS2 you need to use this project: https://github.com/RobotWebTools/ros2-web-bridge As you already are saying rosbridge_server package is ROS1 only.
> 
>     2. duckietown_msgs should be available when you build packages from https://github.com/lgsvl/duckietown2

hey, Martins,

 I don't necessary want to use ROS2, but  building `duckietown_msgs` requires ROS2, right ?
     

## build duckietown_msgs 

by following the build process, but I am not sure how [build-ros2-lgsvl](https://github.com/lgsvl/build-ros2-lgsvl) helps to build, the description is burred. anyway I can built `duckietown_msgs` using `ament build` : 
  
which then create some new folders under  `/duckietown_msgs` : 
 
       1) build 
       2) install , under which has "local_setup.bash,  lib folder, include folder, share folder 
       3) e.t.c

##  how to use the  duckietown_msgs: 

* sourced local_setup.bash in install folder, which I think is to setup this duckietown2 path to the system env,  
     
 *  export /install/lib/python3.x/site_packages to $PYTHONPATH, which I think will help the user defined ros node to understand the message type

*  start simulator and roslaunch  roswebsocket 

but looks it doesn't work in this way, the websocket report cant' advertise the duckietown_msgs type

Can you help to have a clear example here with how to use the duckietown_msgs ?

Thanks again

martins-mozeiko, 2019-05-21:
You should not use build-ros2-lgsvl repository unless  you are building webOS distribution with ROS.

If you are building duckietown_msgs from https://github.com/lgsvl/duckietown2/tree/master/00-infrastructure/duckietown_msgs then that is ROS2 only.

For ROS1 you can use original duckietown repository. Here's what should work to build it and run with rosbridge (using kinetic):

    source /opt/ros/kinetic/setup.bash
    cd duckietown
    catkin_make -C catkin_ws
    source environment.sh
    roslaunch rosbridge_server rosbridge_websocket

I have not used ament, so I'm not sure how exactly to run instead of catkin. But it should be possible to run it in place of catkin_make to run the build.

closed 2019-08-19 by martins-mozeiko