lgsvl/simulator #159: Autopilot feature?

AravindChandradoss, 2019-05-15:
I want to know whether LGSVL has any autopilot or any similar feature? If so, how to enable it?
Coz, I am planning to use LGSVL for some data collection and testing.

luqiang21, 2019-05-15:
@AravindChandradoss LGSVL simulator has the option to enable cruise control (can be enabled on the left panel where you enable sensors) for the ego vehicle. You need to control the steering while collecting data. The other option could be: you can use AD stack like Apollo/Autoware to drive the car and collect data.

AravindChandradoss, 2019-05-15:
Thank you @luqiang21  
I will check that!   
and btw how the other NPCs are driving then? Coz, when I intentionally blocked the path of other spawned cars, they were able to avoid my ego-car and drive.
Thanks

martins-mozeiko, 2019-05-16:
Other NPC vehicles have extra logic to follow annotated map. But the way how they drive is not physically accurate, it only looks realistic.

closed 2019-05-16 by AravindChandradoss