lgsvl/simulator #160: Apollo 3.0 MSF Localization Support

jonghl9, 2019-05-16:
I've recently discovered your simulator, and was thoroughly impressed with Baidu's Apollo support.
Since then, I've been using the simulator to do some experiments with Apollo 3.0.
However, I'm currently stuck with getting the simulator to work with Apollo 3.0 running MSF localization.
The following are the steps I took in an attempt to get MSF working with the simulator.

- I have collected the pcd frames and poses data directly from the simulator manually to create the msf localization map.
- I then created `local_map` using `apollo/scripts/msf_local_map_creator.sh` with the data I collected from the simulator and configured `modules/localization/conf/localization.conf` file to make use of the resulting local map.
- Looking at the localization error logs it looked like there was a timestamp mismatch with the LiDAR, so I corrected that by modifying the code in the simulator and no more error logs were displayed.

Even with the steps I took above, MSF localization won't work with the simulator. 
So my main question is, will this simulator support Apollo's MSF localization?
If the developers had any luck with getting msf to work, can you share your experiences?
Thanks.

andytolst, 2019-05-17:
Hi @jonghl9 
We did try running MSF localization with Apollo and did get it working, but it's not very stable, and it's hard to debug since core algorithm is close-source in Apollo and supplied as pre-compiled binary.

Here are some tips you might find useful:
- If you are running Apollo and Simulator on separate machines - make sure that time on both machines is synchronized, MSF is very strict about timestamps, and if you are running on the same machine, make sure that Simulator is running at least with 25 FPS and you get stable IMU message rate

- For creating localization map it's better to use this approach: collect rosbag with gnss odometry, compensated Lidar and ins_stat, then use `scripts/msf_simple_map_creator.sh` script with the rosbag to create localization map (don't forget to rename / copy resulting lossy_map to local_map). The script needs a folder with rosbag, extrinsics between novatel and Lidar (you can use `https://github.com/lgsvl/apollo-vehicle/blob/master/XE_Rigged/velodyne_params/velodyne64_novatel_extrinsics_example.yaml`) and UTM zone id, that should be 10 for San Francisco. 
You can even try to run Apollo in autonomous mode around the block while collecting rosbag, so you don't have to drive car manually.

- You can use MSF visualization tool to check all 3 components (Lidar localization, GPS and INS): enable flags `--lidar_debug_log_flag=true` `--integ_debug_log_flag=true` `--gnss_debug_log_flag=true` in localization.conf
Run `./scripts/localization_online_visualizer.sh` - this will open a new window with visualization tool, it will be visible only when it's getting lidar data, so make sure you have bridge and simulator running, it should look like this:
[image]

- You can try to play with `--lidar_map_coverage_theshold=0.8` in `localization.conf` - this is Lidar localization threshold (how good is Lidar match) and some other parameters, e.g. take a look at this commit: https://github.com/lgsvl/apollo/commit/2c8588668f7cb5763045eb0848adb7a6c196764a
where we disabled a few checks in Apollo code.

Let us know if you have any questions / issues.
And if you find a way to make MSF work reliably with the simulator - pull requests are always welcome :)

jonghl9, 2019-05-18:
Thank you very much for the detailed explanation :) MSF was hard to debug indeed because of the binaries. I'll make sure to make a pull request if I can get MSF to work reliably.

closed 2019-05-18 by jonghl9

andytolst, 2019-05-21:
@jonghl9 make sure you have this fix for IMU publishing rate: https://github.com/lgsvl/simulator/issues/138

xmyqsh, 2019-06-08:
@andytolst You could try to enable `Apollo.Drivers.Gnss.Heading` to have a test.

Juntu-hub, 2021-06-22:
Hello, @jonghl9 Have you got a stable msf_localization so far? I am also confused about the local_map build by myself recently. The  msf_lidar is always off and on. Could you please share some keypoints?
@xmyqsh Could you please tell me how to enable `Apollo.Drivers.Gnss.Heading`? Is this in lgsvl sensor or apollo's config ? Are you sure the heading is necessary for msf_localization?