lgsvl/simulator #161: Why Some game object of other scene seems blink in the screen

johnny77lee, 2019-05-16:
Hi , 

I have some question about add new scene to project ,
1. when I run the scene "Shalun" from menu , I found some game object seems blink in the screen like the video "screen_blink" 
[screen_blink.zip](https://github.com/lgsvl/simulator/files/3186800/screen_blink.zip)

2. If I add a new scene with simple game object (ex. just terrain & some roads) , the screen seems blink when it run too.
3. But it seems wouldn't occurred with the scene "SanFrancisco" or SimpleMap. 

Would you have any idea to fixed it ?
Would you provide the rule about "what kind of basic objects we need to create a new scene ?" 

Thanks

martins-mozeiko, 2019-05-16:
This is probably z-fighting. I think we need to raise road markings a bit above ground.

When you are adding new objects make sure they are raised a bit above ground. Even 1 millimeter will be enough.

johnny77lee, 2019-05-17:
Hi Martins ,

Thanks for your comment ,.
But in the Scene "Shalun" , the blink & normal road line seems in the same object "Road Mesh"
It seems couldn't raise a bit above ground for abnormal road mark . 
And it seems have different result with different View Rotation even in edit mode like video "RoadMesh_LaneMark_Blink".
[RoadMesh_LaneMark_Blink.zip](https://github.com/lgsvl/simulator/files/3190609/RoadMesh_LaneMark_Blink.zip)

Would you provide the rule about "How to modify the abnormal road mark in the Object"RoadMesh" ? "

In the another video (another_blink) , the crosswalk mark in the far way seems blink too,
Is it still cause by "z-fighting" ?
[another_blink.zip](https://github.com/lgsvl/simulator/files/3190612/another_blink.zip)

Thanks

martins-mozeiko, 2019-05-17:
Yes, all this is z-fighting. While the lane markings are inside the road texture, the sidewalks and arrows are separate objects. You can move them tiny bit up.

I'm not sure why lane markings are blinking inside Unity Editor for RoadMesh, it may be that editor is rendering with lowered quality for performance reasons. But all other videos are definitely showing z-fighting.

johnny77lee, 2019-05-20:
Hi Martins ,

If all of lane line blink is cause by Z-fighting , why just the lane line in some area would be blink in the screen(see the below image & video
[LaneLine_blink_Shalun.zip](https://github.com/lgsvl/simulator/files/3196224/LaneLine_blink_Shalun.zip)
)
[image]

In the Video , we could found (0:00 - 0:18) the vehicle in the "Lane Line  blink area" , we could see the lane line blink in the screen . But after we leave the area (after 0:18) the Lane Line wouldn't blink in the screen .

As I know , All of Lane Lines are in the same object and it use the same mesh file "RoadMesh1" .
So.....I don't understand why just some lane line blink in the screen ?
Could you provide more detail about how the z-fighting cause just some Lane Line blink ?

And Would you provide the rule about how to fixed the Lane Line blink ?
(maybe need app or unity tools ? )

Thanks

johnny77lee, 2019-05-22:
Hi @martins-mozeiko ,

About my last comment , would you have ant idea about "Lane Line blink area" .
And If I want to fixed it in my local side , would you have any simple rule to do it ?

Thanks

martins-mozeiko, 2019-05-30:
The objects blink because there are multiple meshes in same place. When polygons are parallel to each other and occupy exactly same coordinates then due to float precision errors pixels for polygon are sometimes on top, sometimes on bottom. Wikipedia has good illustration for this on [Z-fighting](https://en.wikipedia.org/wiki/Z-fighting) article: https://en.wikipedia.org/wiki/Z-fighting#/media/File:Z-fighting.png

It seems you are using older version of Shalun map. You should check how it looks in master. We have adjusted the map to have proper geometry so no "blinking" is happening anymore.

closed 2019-06-12 by johnny77lee