lgsvl/simulator #162: How to fully utilize gpu lidar

bigsheep2018, 2019-05-16:
Hello lgsvl team @martins-mozeiko,

I am writing to check about how to fully utilize GPU for lidar computation (raycast i guess) or do I miss any lidar configuration setup for GPU-based lidar?
I checked the GPU situation while I am running the newly released simulator (v2019.04). It seems the GPU is not fully loaded during runtime. In fact, I found the only difference is 100mb+ more occupation of GPU memory while I turn on a 128-lidar.  
Besides, the 128-lidar is quite slow in Unity Editor game mode.
Please check the following pics for the GPU situation on my desktop while running the simulator:

1) without lidar
[image]

2) with lidar (128 channel)
[image]

If possible, please give more details if there is a way figure this problem out (I am quite new to Unity3d and c#).
Thanks in advance.

martins-mozeiko, 2019-05-16:
The lidar implementation we have is not fully optimized. While currently it works well for lower ray & measurement count, it requires too much computations on higher numbers. We plan to optimize it, but best you currently can have for real time lidar simulation is around 32 rays.

opoppo, 2019-06-08:
Seems that in Editor game mode it will always be 3 fps with lidar on, but after complied into standalone executable fps would be improved significantly.

martins-mozeiko, 2019-08-19:
With have greatly optimized lidar simulation performance in our new simulator [2019.07-rc1](https://github.com/lgsvl/simulator/releases/tag/2019.07-rc1) release. It is able to run 128 ray lidar simulation realtime.

If you have any more questions on this please open a new issue.

closed 2019-08-19 by martins-mozeiko