lgsvl/simulator #165: Point cloud intensity on glass and synthetic perturbation?

opoppo, 2019-05-21:
[image]

I notice that synthetic point clouds from the simulating LiDAR have two major gaps with real-world point clouds as shown in the picture above. LEFT synthetic data from simulator, RIGHT real-world data from a Velodyne 64E-S2.

1\ laser intensity from glass materials (typically windshields and windows along the streets) should be none or zero? as in reality, lasers don't actually reflect from glass materials as if they don't exist.

2\ the laser points seem to be a bit dull and static, since real point clouds always come with much perturbation. So I suggest adding some random or gaussion noise to the measurements to better simulate.   

Is there any suggestion on these?

opoppo, 2019-05-21:
or any direct ways that I can modify these on my own?

martins-mozeiko, 2019-05-21:
We are currently generating lidar intensity from color texture of object. And transparency of surface is determined from alpha channel of texture. Other vehicles currently do not have windows set as transparent - that's why lidar "sees" them as nontransparent surfaces.

If you want to get transparency for other vehicles, you will need to modify texture for these cars to have alpha channel for window parts. For example, for truck the texture is this file: https://github.com/lgsvl/simulator/blob/master/Assets/Textures/DeliveryTruck/TruckCockPit_Black.jpg

For example, the direction arrows on road are separate objects with transparent texture: https://github.com/lgsvl/simulator/blob/master/Assets/Textures/SF_RoadArrows.png (you won't see anything, because arrow color is white)

This is how it looks like in simulator (I placed fake direction arrow in scene): [image]

The geometry is square, but lidar hits only non-transparent parts.

closed 2019-05-22 by opoppo