lgsvl/simulator #167: dose lgsvl support deep reinforcement learning?

Derekabc, 2019-05-23:
Hi, I am new to LGSVL, is it possible for me to build a reinforcement learning model using LGSVL, and any suggestions? Do i need to build a customized gym environment to intereact with LGSVL?  I appreciate any suggestions and insights!

hadiTab, 2019-05-23:
@Derekabc it is possible to train an agent in the simulator using reinforcement learning. The Python API provides this possibility. We will be releasing a gym environment very soon as well.

hadiTab, 2019-05-29:
@Derekabc we've released a gym environment for the LGSVL simulator. It's available [here](https://github.com/lgsvl/gym-lgsvl).

Derekabc, 2019-05-29:
Cool------------------ Original ------------------From: Hadi Tabatabaee  <notifications@github.com>Date: Wed,May 29,2019 4:03 PMTo: lgsvl/simulator <simulator@noreply.github.com>Cc: Jay Chen <chendondon19930606@gmail.com>, Mention <mention@noreply.github.com>Subject: Re: [lgsvl/simulator] dose lgsvl support deep reinforcement learning?(#167)@Derekabc we've released a gym environment for the LGSVL simulator. It's available here.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/lgsvl/simulator/issues/167?email_source=notifications\u0026email_token=AGETZR3SBFIWEMDJSXOESMTPX3OSNA5CNFSM4HPHGXG2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODWQPRMY#issuecomment-497088691",
"url": "https://github.com/lgsvl/simulator/issues/167?email_source=notifications\u0026email_token=AGETZR3SBFIWEMDJSXOESMTPX3OSNA5CNFSM4HPHGXG2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODWQPRMY#issuecomment-497088691",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

closed 2019-06-13 by martins-mozeiko