lgsvl/simulator #170: Is there a pedestrian path editor in Map Tool Util Editor?

s894330, 2019-05-27:
Hi 

In scene sanfrancisco, there are pedestrians path in map which using the script "map pedestrian segment builder", I plan to make the same pedestrians path in scene "simplemap", but I can not find any tool in "MapToolUtilEditor" to let me do this job just like the Lane waypoint creation using "MapLaneSegmentBuilder" (create temp map waypoint, then make lane)

I wonder is there any tool to make pedestrians path build easily? just like lane waypoint do

thank you

EricBoiseLGSVL, 2019-05-28:
@s894330 ,

Hello.  Not currently.
To make a new path:
1) enable waypoint visiblity
2) enable handle visiblity on selected pedestrian builder
3) increase the local waypoint array
4) increase the x or z value for each local point to offset so you can use the handles to position
5) place as needed. The editor buttons on the pedestrian builder class work but not the ones in the map tool.
You can also make one a prefab and just duplicate for the simple map.

-Eric

s894330, 2019-05-31:
hi Eric

thanks for your help
I will try it later :-)

closed 2019-05-31 by s894330