lgsvl/simulator #171: Simulation with Apollo3.5

zezhang, 2019-05-27:
Hello LGSVL Team,

I understand that the LGSVL Simulator currently supports the weather condition and can integrated with apollo3.5. Here are my questions:

1) Does the simulator actually simulate the planning trajectory as well as the control command generated from Apollo (or it only takes the perception information from Apollo)?

2) Does the weather condition actually affect the driving behavior of the car? (e.g. If it's raining, the breaking distance should be longer). If it does, where does the driving behavior get modified (Inside apollo control/planning logic or inside the simulation engine)? 

Any help would be appreciated!

Thanks

martins-mozeiko, 2019-05-28:
1. Simulator does not generate any kind of planning or control commands. It only provides sensor information (camera image, lidar point cloud, imu, gps) to Apollo. And apollo processes this data and gives back control command.

2. Currently weather effects impact only visuals - camera image.

zezhang, 2019-05-28:
Thanks for the quick response!

closed 2019-05-28 by zezhang