lgsvl/simulator #173: Change the bridge port

daohu527, 2019-05-28:
The lgsvl/apollo-3.5 provide a bridge, which listens on port 9090 in default. Then I change the port to 9098 because the port in linux is occupied. The problem is that I can't find any place to change the port in the simulation.  
Could you help me to deal with that ?

xmyqsh, 2019-05-28:
@daohu527 
`
Assets/Scripts/ROSAgentManager.cs:100:            var port = PlayerPrefs.GetInt($"ROS_AGENT_{i}_PORT", 9090);
`
`
Assets/Scripts/RosBridgeConnector.cs:25:    public const int DefaultPort = 9090;
`
change the code and recompile.

Alternatively,
Run simulator and apollo in different machine, and forwarding the port
`
ssh -L 9090:localhost:9098
`

daohu527, 2019-05-28:
@xmyqsh Thx
I see a little about the apollo bridge code. Is there any conf in python? The bridge is translate the python message. I don't want to compile the project, or maybe later I will try.  

By the way in my case, I use a server run apollo3.5 and change the port to 9098, then open the simulation in my localhost. Have can I set the "ssh -L 9090:localhost:9098", which need a server name ?
I looked at a similar tutorial, but it was a bit confused. Where do I run the cmd, and I try "ssh -L 9090:localhost:9098 localhost" It doesn't work.

martins-mozeiko, 2019-05-28:
No need to change code in simulator to use different port. Just enter it directly in address:

[image]

To make bridge to listen on different port, edit `scripts/bridge.sh` and append `-port 9098`

daohu527, 2019-05-29:
@martins-mozeiko  Thx for you reply!
 I try to add the port in "192.168.1.102:xxxx", seems the input is not allowed to be so long? I will try again!

martins-mozeiko, 2019-05-29:
There should not be any limit on length there. Maybe you have resized window too small? It probably does not fit on screen, but it actually stores full value. Try moving cursor left and right to see if input is scrolling for full value.

daohu527, 2019-05-29:
@martins-mozeiko 
You are right, It work well by "192.168.1.102:xxxx", maybe some problem in “Shif” key and make some garbled in the input box. 
Thank for you job!

closed 2019-06-03 by daohu527