lgsvl/simulator #174: Vehicle Calibration on XE_Rigged-apollo/LG_SVL_1080

xmyqsh, 2019-05-28:
I'm gone through and compared the [vehicle prefab](https://github.com/lgsvl/simulator/blob/master/Assets/Prefabs/AgentCandidates/XE_Rigged-apollo.prefab) in XE_Rigged-apollo  and vehicle [parameters](https://github.com/lgsvl/apollo-vehicle/blob/099594c780788a491b99bd2ef6f6a3ad4a8e1ee9/LG_SVL_1080/vehicle_param.pb.txt) and [calibration](https://github.com/lgsvl/apollo-vehicle/blob/099594c780788a491b99bd2ef6f6a3ad4a8e1ee9/LG_SVL_1080/calibration_table.pb.txt) in LG_SVL_1080.

I found there are lots of mismatched parameters between XE_Rigged-apollo and LG_SVL_1080.

For example, the [mass](https://github.com/lgsvl/simulator/blob/master/Assets/Prefabs/AgentCandidates/XE_Rigged-apollo.prefab#L3932) and the [center of mass](https://github.com/lgsvl/simulator/blob/master/Assets/Prefabs/AgentCandidates/XE_Rigged-apollo.prefab#L4103) vs [mass on wheels](https://github.com/lgsvl/apollo-vehicle/blob/099594c780788a491b99bd2ef6f6a3ad4a8e1ee9/LG_SVL_1080/calibration_table.pb.txt#L15-L18); [max steer angle](https://github.com/lgsvl/simulator/blob/master/Assets/Prefabs/AgentCandidates/XE_Rigged-apollo.prefab#L3982) vs **[wheel base](https://github.com/lgsvl/apollo-vehicle/blob/099594c780788a491b99bd2ef6f6a3ad4a8e1ee9/LG_SVL_1080/vehicle_param.pb.txt#L16) and [min turn radius](https://github.com/lgsvl/apollo-vehicle/blob/099594c780788a491b99bd2ef6f6a3ad4a8e1ee9/LG_SVL_1080/vehicle_param.pb.txt#L10)** vs _[max_steer_angle](https://github.com/lgsvl/apollo-vehicle/blob/099594c780788a491b99bd2ef6f6a3ad4a8e1ee9/LG_SVL_1080/vehicle_param.pb.txt#L13) and [steer_ratio](https://github.com/lgsvl/apollo-vehicle/blob/099594c780788a491b99bd2ef6f6a3ad4a8e1ee9/LG_SVL_1080/vehicle_param.pb.txt#L15)_; and other undetermined physical parameters, eg. [cornering stiffness](https://github.com/lgsvl/apollo-vehicle/blob/099594c780788a491b99bd2ef6f6a3ad4a8e1ee9/LG_SVL_1080/calibration_table.pb.txt#L13-L14) which should depand on the mass and the steer friction and so on.

Despite these, only the `longitude_controller_conf` has been calibrated(difference from the apollo mkz default config).

My big doubt is on lateral control: The max steer angle in XE_Rigged-apollo is 39.4, while we can also calculate it from [LG_SVL_1080](https://github.com/lgsvl/apollo-vehicle/blob/099594c780788a491b99bd2ef6f6a3ad4a8e1ee9/LG_SVL_1080/vehicle_param.pb.txt#L10) which is less than 30. There is 25% error between them. But the lateral_controller_conf has not been calibrated(same as the apollo mkz default config).

The parameters I mentioned above affect both control and planning model.
Does the vehicle run as expected?  Or the vehicle could run better and more precisely. Or there is some secret in the implementation of the vehicle model in the simulator.

daviduhm, 2019-05-28:
LG_SVL_1080 is outdated and is no longer available (we removed it from our repo on March 29). Please use [XE_Rigged](https://github.com/lgsvl/apollo-vehicle/tree/master/XE_Rigged).

For a lateral control, you are right and the max steering angle affects control module. We have calibrated it and updated the max steering angle as 39.4 and some other parameters as well (mass, steer ratio, wheel base, etc) for better/precise right turn on intersections. These updates are in our internal repo and will be pushed to our github this week. In the meantime, you can use XE_Rigged for running Apollo without any major issues.

xmyqsh, 2019-05-29:
@daviduhm Look forward!
Honestly, I have not run your simulator. I will try it the following days.

xmyqsh, 2019-05-29:
@daviduhm What's the map coordination and vehicle coordination the simulator use?

daviduhm, 2019-05-29:
If you are asking about UTM coordinates of our SanFrancisco map, the map origin is `10U 552874 4182486` by default. Also, I'll try to ping in this thread once the new vehicle calibrations are released.

daviduhm, 2019-06-03:
@xmyqsh Hi, I just wanted to let you know that we have pushed the vehicle calibrations including updated lateral control parameters in this [repo](https://github.com/lgsvl/apollo-vehicle). Please pull the latest submodules in your Apollo clone and try our simulator. Thanks!

xmyqsh, 2019-07-12:
@daviduhm 
Is XE_Rigged the one used in apollo-3.5 XE_Rigged_Apollo3.5?

closed 2019-07-17 by xmyqsh