lgsvl/simulator #176: echo lidar data

ErijBNT, 2019-05-28:
when I run rostopic echo of the lidar topic I don't get the data.  Why is that?
[image]

martins-mozeiko, 2019-05-28:
Can you explain more - what are you running? Which map and vehicle are you selecting?

It seems that simulator is sending lidar data, but it has no points detected... Where is vehicle on the map positioned?

ErijBNT, 2019-05-28:
@martins-mozeiko I am using the lgsvl vehicle and the San Francisco city.  Thelidar sensor is enabled and I can visualise the points in the simulator when enabling the sensor effects.

martins-mozeiko, 2019-05-28:
LGSVL vehicle expects ROS2 bridge & ROS2 environment. You are running rviz from ROS2, right?

ErijBNT, 2019-05-28:
@martins-mozeiko No I am running it from ros1. In fact,  I was able to get the pictures saved such as the lane following project but ina ros 1 environment

martins-mozeiko, 2019-05-28:
Depending on message type some messages are exactly same between ROS1 and ROS2. But some messages are formatted differently in rosbridge protocol. You need to use ROS2 bridge and ROS2 environment if you want to use LGSVL vehicle.

ErijBNT, 2019-05-28:
@martins-mozeiko Well,  I used the ROS 2 environment but the rosbridge doesn't transport images. I always get black pictures while trying to save it.

martins-mozeiko, 2019-05-28:
Simulator supports sending images over ROS2 just fine. This is how our [lanefollowing](https://github.com/lgsvl/lanefollowing) project gets the image - check the video. You'll see captured image displayed in new window (by ROS2) node.

How exactly are you capturing image? Note that we are always sending jpeg compressed image, so you might need to decompress it if you want RGB values.

ErijBNT, 2019-05-28:
@martins-mozeiko  I did the same as the lane following project but the images are black

martins-mozeiko, 2019-05-28:
Please check if there are any errors in output when docker container runs. And check logfile for Simulator (it is in `~/.config/unity3d/LG Silicon Valley Lab/Automotive Simulator/Player.log` location).

ErijBNT, 2019-05-28:
Thank you for your help. 
I have another question I don't if is it feasible.  Can I process the data in real time and apply object classification and labelling and observe it in the simulator so that the vehicle can detect traffic lights

martins-mozeiko, 2019-05-28:
This kind of detection is happening already now in Apollo or Autoware. They are receiving images over ROS or Cyber, doing detection real-time, and using result to control the vehicle so you can see whether it moves or not in simulator.

ErijBNT, 2019-05-29:
Hello, 
I faced the same problem and using ros2
[image]

martins-mozeiko, 2019-05-29:
Have you tried writing ros2 node that uses this lidar topic to actually see if there is any data? Maybe "ros2 topic echo" does not want to output large binary data to terminal?

When I'm using ros2 ardent version then it shows this:
```
data: !!binary ""
```
But the data is still there.

ErijBNT, 2019-05-29:
When I run ros 2 topic echo for the apollo vehicle 
There is an output. And the data shows

martins-mozeiko, 2019-05-30:
Is there anything is bridge output? warnings/errors?

What about in log file for simulator?

ErijBNT, 2019-05-30:
I run the SIMULATOR  as a standalone executable.

ErijBNT, 2019-05-30:
@martins-mozeiko  There is no errors of the output of the bridge.
and this is the player.log file
[Player.log](https://github.com/lgsvl/simulator/files/3238546/Player.log)

martins-mozeiko, 2019-05-30:
You're right. we have a bug for ROS2.
You can fix it by removing this line: https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/LidarSensor/LidarSensor.cs#L618
And putting following two lines there instead:
```
Array = RosPointCloud,
Length = count * 32,
```

ErijBNT, 2019-05-30:
Thanks a lot! it works now.
[image]

closed 2019-06-13 by martins-mozeiko