lgsvl/simulator #179: How do you change it?

ehduk90, 2019-05-30:
[image]

Hi, how's it going?

I was trying to use OP_planner on the LS simulator and found the data with the problem as above.

How do we modify this?
Or can you provide modified data?

ehduk90, 2019-06-03:
I tried to change this using autoware tool.

This work seems too inefficient.

Can you provide me with the modified file?

luqiang21, 2019-06-03:
@ehduk90 Are you using the latest simulator code and [autoware-data](https://github.com/lgsvl/autoware-data)? This problem was due to a wrong commit merged. We already reverted it. And you can also find the file from autoware-data repo.

ehduk90, 2019-06-04:
Thank you for your response.

I used the simulator and autoware_data that I recently received, as you mentioned, and found the above problems.

After seeing your answer, I received autoware_data again and confirmed that the same problem remains.

The map I used is a simple map, and I used it as follows.
[image]
[image]
[image]

alva786, 2019-06-18:
Hi , We are aslo using the latest simulator code and autoware-data(vector_map_sim_shalun), and the same result, the center line road network as attached picture, Any suggestion??

[image]

martins-mozeiko, 2019-06-18:
@ehduk90 @alva786 
Which commit autoware-data are you using? lgsvl/autoware-data@3a51cd48a1726f681605248f993bf27282a24b8e commit (which is latest) should work fine and have correct vector map for SimpleMap and Shalun maps.

@ehduk90 The way how you have set up Autoware Runtime Manager is a bit different how we use it. We enable Map, Sensing, Localization, Detecting, Mission Planning, Motion Planning on first tab ("Quick Start"). We do not enable anything under "Map" tab. Maybe try doing this.

alva786, 2019-06-19:
@martins-mozeiko
Yes I use lgsvl/autoware-data@3a51cd4 commit. 

For Shalun ,
I just modify the lane.csv in Shalun maps. At LnID:3965 , modify the FLID from 3966 to 0.
Because original lane.csv make "op_globle_planner" process died during Construct RoadNetwork(V2)

For SimpleMap, 
It's normal when planner construct roadnetwork, process not die.
So, I use the original vectormap file, and the VectorMap CenterLines is also abnormal like attached picture.
[image]

martins-mozeiko, 2019-06-19:
And how did you open this rviz? Did you click only buttons on first tab (Quick Start) or you used different way?

alva786, 2019-06-19:
@martins-mozeiko   , Abnormal when use op_globle_planner (not use way_planner in quick start)

ehduk90, 2019-06-20:
@alva786 

Does your virtual car work normally?

I want to move the car with this function. However, my virtual car does not move and my steering is reversed even when I move to cruise control.

martins-mozeiko, 2019-06-20:
@alva786 can you explain what we need to do to enable op_global_planner? Which buttons/checkboxes do we need to enable in Autoware Runtime Manager?

alva786, 2019-06-21:
@martins-mozeiko 
op_global_planner can do "lane change" function in global path planning, like as video https://www.youtube.com/watch?v=jmWaE1CzBas
checkbox : as attached image 
[image]

alva786, 2019-06-21:
@ehduk90
Thanks for your share.
You can try with motion by dp planner,
I also cannot move car with  'pure_pursuit'.

martins-mozeiko, 2019-06-21:
But what other checkboxes do I need to enable? From this screenshot I see that you have checked vel_pose_connect. But what else? Can you provide list of all actions I need to do after I run `autoware/ros/run` script?

alva786, 2019-06-21:
@martins-mozeiko
After autoware/ros/run
(1)In Map page, Load Vector map (autoware-data/data/map/vector_map_sim_simpleMap/) & TF
(2)In Computing page, click op_global_planner
(3)Open Rviz and see Vector Map CenterLines
Now you can see the VectorMap CenterLines is abnormal .

martins-mozeiko, 2019-06-21:
@alva786 Thank you! I was able to reproduce this. We'll check how we can fix this.

hatem-darweesh, 2019-08-13:
To fix this problem, you need to force OpenPlanner to use vector map parser version 1. 
in op_global_planner_core.cpp comment the version check condition (m_MapRaw.GetVersion()), 
use only "PlannerHNS::MappingHelpers::ConstructRoadNetworkFromROSMessage" to create the map. 
for the local planner to work fine too, please modify the same in any place that construct the map. 

- The reason for this problem is that these maps are not created with the same library as the original Aisan Tech. maps.

closed 2020-04-07 by EricBoiseLGSVL