lgsvl/simulator #181: Can't find tutorial or codes for "enable traffic" and "enable pedestrian" functions

dongyy1148, 2019-05-31:
Hi,
The simulator has the functions "enable traffic" and "enable pedestrian" to add random npc vehicles and pedestrians in the map. Besides, these npc vehicles have their own driving routes and are able to avoid collision. I want to learn how the simulator realizes these functions, however, I cannot find any introductions in the documentation or the codes. Could you help introduce how you realize these functions? Thanks!

martins-mozeiko, 2019-05-31:
If you open Window -> Map Tool Panel in Unity Editor and click "Show/Hide Map" button you will see extra lane annotations in scene. NPCs use these annotations to know where to drive, and how to cross intersections. And Pedestrians use annotations on sidewalk to know where to walk.

If you click on individual lane markings you will see extra parameters in inspector that you can set. For example - lane type (white, yellow, double yellow), turn information, etc...

dongyy1148, 2019-06-01:
Thanks! I can find the npc and pedestrians in the unity editor now. But I still can't get why they will stop if they meet the obstacles.

martins-mozeiko, 2019-06-03:
Avoiding obstacles for NPC cars is done by doing forward raycasts here: [NPCControllerComponent.cs#L1605](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/NPCControllerComponent.cs#L1605).

Based on the results NPC will either change lane - [NPCControllerComponent.cs#L1004](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/NPCControllerComponent.cs#L1004) or will make passing maneuver - [NPCControllerComponent.cs#L1058](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/NPCControllerComponent.cs#L1058)

For pedestrians avoiding other pedestrians is controlled by `avoidancePriority` variable in [PedestrianComponent.cs#L136](https://github.com/lgsvl/simulator/blob/d4cd16ecde4c6c42284d136eea19d45e27a4e662/Assets/Scripts/Driving/PedestrianComponent.cs#L136) When two pedestrian collide, the one with lower priority will turn away. This is controlled by Unity's built in [NavMeshAgent](https://docs.unity3d.com/ScriptReference/AI.NavMeshAgent.html) functionality.

dongyy1148, 2019-06-07:
Thanks for the reply! Is this function built in the Unity editor or added by your simulator? Which means that if I add one new map in the Unity, will the NPC have the avoidance function?

martins-mozeiko, 2019-06-07:
This is custom code we have implemented in our simulator. It will automatically work for new maps as long as your have lane annotations.

dongyy1148, 2019-06-07:
Gotcha. Thanks a lot for your help!

closed 2019-06-11 by dongyy1148