lgsvl/simulator #184: Simulator & Apollo 3.5 connecting but not running route

ninjacoder0, 2019-06-05:
[image]

I have downloaded and successfully ran Apollo 3.5 and your simulator. 
I have ticked the right modules then as the page says, I clicked on "route editing" followed by selecting a destination by clicking on a lane line and clicking "Submit Route".
Although the vehicle would not move. When I use the arrow key on the keyboard it does move on the Apollo page then though.

Why is not not moving autonomously? 
Thank you for your help.

martins-mozeiko, 2019-06-05:
Can you try toggling control module off and on again?
You can also launch `cyber_monitor` inside docker to see what messages are being published by Apollo. Check what is being sent in control message.

ninjacoder0, 2019-06-10:
[image]

I have switched on all the modules on Apollo as the website instructs and the simulator has the correct modules switched on too. Please see the cyber_monitor above.  What am I doing wrong for it not to follow the route instructed? Like the first comment, the vehicles moves with the arrow keys and so does the Apollo page, but does not automatically do it following the route.

Thank you

luqiang21, 2019-06-10:
@ninjacoder0 It looks like your perception is not running since no messages under `/apollo/perception/obstacles`. In Apollo 3.5, perception and prediction modules should be running first, then planning will be able to run. Then you will get the planning trajectory which is not currently shown in your screenshots. I suggest you to run the perception module using command line to see what is happening to perception. 
Run `mainboard -d modules/perception/production/dag/dag_streaming_perception.dag` inside docker.

ninjacoder0, 2019-06-11:
[image]

When I run the command given, I get this result back. What would you suggest I do?

luqiang21, 2019-06-11:
@ninjacoder0 You are running the wrong command. There is no dag folder under `perception`. It is better to just copy and paste the command.

ninjacoder0, 2019-06-11:
[image]

Arghh! Apologise, I am using two PC's.
Anyways, corrected and bringing up errors?

luqiang21, 2019-06-11:
@ninjacoder0 It looks like it is not a problem of LGSVL simulator. We didn't experience a similar problem like this within our team. You can check issues on Apollo repo. One possible solution is [here](https://github.com/ApolloAuto/apollo/issues/6306). Hope it can be a help for you.

martins-mozeiko, 2019-06-11:
@ninjacoder0 Make sure you have [nvidia-docker](https://github.com/NVIDIA/nvidia-docker) installed before running dev-start.sh. Our Apollo 3.5 fork uses nvidia-docker so it is easier to access GPU from docker. You can test if this works by running `nvidia-smi` command inside Apollo docker container. If it does not show any error, then it is working correctly.

ninjacoder0, 2019-06-12:
@martins-mozeiko @luqiang21 
Thanks for both your help.
My  nvidia-docker was working, even 3.0 worked fine for me... But I reinstalled it anyways and it worked!! Thank you both :)

closed 2019-06-12 by ninjacoder0

Iqbalparvi, 2019-09-28:
@ninjacoder0 I have facing  exactly the same problem as you discuss above that "the perception and traffic light modules automatically off " in dreamview and there is no obstacle appear the only Ego car move that can manually control from lgsvl .can you please give some more details how you overcome this issue?
Regards