lgsvl/simulator #185: Sunnyvale Photogrammetry Feature?

brac45, 2019-06-05:
Hello,

The project has impressive integration for Apollo and it's very great.
I was reading the overview document for the simulator, and noticed an interesting feature which supposedly shows Sunnyvale using photogrammetry (https://www.lgsvlsimulator.com/downloads/LGSVL-Simulator-Overview.pdf).
However, I was unable to find such feature in the latest release (April) and pre-release (2019.05-rc1). Is the feature implemented? Thanks!

zelenkovsky, 2019-06-06:
Dear Jongho,

  You are right, we are currently working on new simulator release which approximately is going to be available by the end of this month (June 2019). New release will include a lot of new features where one of them is a support for new Unity3D HDRP rendering pipeline. Image you've seen from the PDF file is rendered using this new version.

Most mysterious question for me - how could you guess by this small picture that it is a part of Sunnyvale? ;)

BR
Dmitry Z.

martins-mozeiko, 2019-08-19:
We have our new release out: [2019.07-rc1](https://github.com/lgsvl/simulator/releases/tag/2019.07-rc1). It includes [Borregas Avenue](https://content.lgsvlsimulator.com/maps/borregasave/) map which  was created using photogrammetry.

If you have any more questions feel free to open new issues.

closed 2019-08-19 by martins-mozeiko