lgsvl/simulator #188: Python examples

faabdal, 2019-06-07:
[image]

I have created my own road environment and when I try to run python examples e.g. 10-npc-follow-the-lane.py , I always get " Exception: Object reference not set to an instance of an object at @0 " (Screenshot above). 

I can run scripts without any problems on San Francisco example but it give me the error when I use my own road. I guess that it some setting in Unity editor. Does anyone know where the problem could be?

Thank you very much

martins-mozeiko, 2019-06-13:
Please check the Unity log file to see what error message does it have when you run this script.

You can find log file in `~/.config/unity3d/LG Silicon Valley Lab/Automotive Simulator/Player.log` location.

faabdal, 2019-06-21:
Thank you for the reply. I have checked the log files but there is not much difference compare to example of San Francisco. There are some missing libraries and components but it is similar to other examples.
Thank you
[Player.log](https://github.com/lgsvl/simulator/files/3312955/Player.log)

martins-mozeiko, 2019-06-21:
Would it be possible for you to run same script with Simulator running in Unity Editor? Then it will show exact filename / linenumber where error happens. Currently it just shows "@0" which does not help to determine location where error happens...

If you are creating your own map, do you have it annotated? If you don't have lane annotations then npc will not be able to follow lanes if they don't exist. Do NPCs drive on your map when you enable them in simulator GUI ("Enable traffic")?

faabdal, 2019-06-26:
You are right, my annotation wasn't done properly and annotation for pedestrians was missing completely. It wasn't clear whether annotation is required by pedestrians or not. 
Thank you for your support and hints.

closed 2019-06-26 by faabdal