lgsvl/simulator #190: Create new a small city map like San Francisco

dongyy1148, 2019-06-07:
Hi,
I have tried different simulations using the San Francisco map connecting your simulator with Apollo, and the results are great. Our company want to try simulations in our own city's map which is not included in your simulator， so we need to create a new map for our city which is able to be used in your simulator and Apollo.
I have read your documents about creating maps. Based on my understanding, we need to first create my city's map in Unity editor, and then annotate the map in the unity, and finally output the map with right format for Apollo. Is my understanding right?
Besides, I think creating map is not an easy work, so I want to ask that do you provide paid service to make a customized map or annotate the map? Thanks a lot!

zelenkovsky, 2019-06-11:
Hello Zheng,

  Thanks for using our Simulator. You are right, we expect autonomous developers using pre-created environments with HD map annotation. HD Map can be exported to Apollo or if company already has HD map, we will add a feature to import HD map into existing scene and align environment with HD map.

 We understand that creating high quality environments is a tedious task, this is why we plan to offer services (via our mapping partners) to create new maps on demand. Maps will vary between small areas with super high fidelity (digital twins) to vast parts of the city precedurally generated.
 
 Next Simulator release will offer special pipeline to bake and host maps independently from simulator itself.

daohu527, 2019-07-26:
Can refer to #199

closed 2020-04-07 by EricBoiseLGSVL