lgsvl/simulator #191: Error with bridge.sh using Apollo 3.5

vatna, 2019-06-08:
I've been trying to run the simulator with Apollo 3.5 and when I execute bridge.sh, I get the following error message:
/apollo/scripts/bridge.sh: line 6:  8036 Illegal instruction     (core dumped) bazel-bin/cyber/bridge/cyber_bridge

Any ideas on what I'm doing wrong? One error message I also see in the previous step (I don't know if it's related or not) is that when I execute bootstrap.sh, it says:
Could not launch module monitor. Is it already built?

martins-mozeiko, 2019-06-11:
Can you run cyber_bridge under gdb debugger? And when it crashes and stops, what location does it show? Check the `bt` and disassembly (few instructions from current location).

vatna, 2019-06-11:
I believe that in order to run gdb, I need to recompile using -g option, but I'm not sure what .c to recompile. Would you be able to provide some more detailed steps to do this?

I don't know if this is relevant info but during the Apollo build, I get the following warnings:
/usr/bin/ld: warning: libfastrtps.so, needed by bazel-out/local-dbg/bin/_solib_k8/_U@ros_S_S_Cros_Ucommon___Uexternal_Sros_Slib/libroscpp.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libfastcdr.so, needed by bazel-out/local-dbg/bin/_solib_k8/_U@ros_S_S_Cros_Ucommon___Uexternal_Sros_Slib/libroscpp.so, not found (try using -rpath or -rpath-link)

I've tried getting this to work on a clean install of both Ubuntu 18.04 as well as 16.04, and get the same error.

martins-mozeiko, 2019-06-11:
Adding -g is not needed. When you run "apollo.sh build_gpu" the output binaries will include debug symbols.

What do you mean by clean install of Ubuntu? Apollo needs to be run as docker image. Host environment does not matter much. Are you building & running Apollo without docker on your host?

vatna, 2019-06-11:
By clean install, I mean it is a new installation onto the hard disk. I then follow the instructions from https://www.lgsvlsimulator.com/docs/apollo3-5-instructions/, which of course includes installing nvidia driver, docker, etc. So when I run `./apollo.sh build_gpu` it is of course done when I am inside the docker container.

I attach a screenshot of gdb.
[image]

vatna, 2019-06-11:
By clean install, I mean that it is a new installation onto the hard disk. I then follow the instructions on https://www.lgsvlsimulator.com/docs/apollo3-5-instructions/ which includes installing the nvidia driver, docker, etc. When I run `./apollo.sh build_gpu` it is when I am inside the container.

I attach a screenshot of gdb.
[image]

vatna, 2019-06-11:
I have 3 cyber_bridge files and I ran the first one earlier - not sure if it was the correct one.
[image]

martins-mozeiko, 2019-06-11:
I think you need to run `./local-dbg/bin/cyber/bridge/cyber_bridge` one.
When you get SIGILL, please run `x/10i $pc` to see which instruction is exactly illegal.

What CPU do you have? Apollo by default assumes that you have CPU with AVX2 instructions. If you do not have AVX2, then you need to disable avx2 instructions when building apollo. You can do this in `apollo.sh` script. Look for `DEFINES="${DEFINES} --copt=-mavx2"` line and either change it to `-mavx` to use AVX instead of AVX2. Or comment it out to completely disable AVX and AVX2 instructions.

vatna, 2019-06-12:
It works now. My CPU didn't support AVX/AVX2 instructions.
Thanks for the support Martins!

closed 2019-06-12 by vatna