lgsvl/simulator #192: odometry data

ErijBNT, 2019-06-08:
I am using the LGSVL vehicle. I am trying to do a SLAM and in order to localize the car, I need the odometry data. how can I get it?

martins-mozeiko, 2019-06-11:
You will need to edit code to publish necessary information.
I suggest to look at existing implementations of IMU or GPS sensors. They are used to collect and publish velocity & acceleration information which probably is very similar to data you need.

ErijBNT, 2019-06-11:
Where to find this implementation ?

martins-mozeiko, 2019-06-11:
GPS is in [GpsDevice.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/GpsDevice.cs) file and IMU is in [ImuSensor.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/ImuSensor.cs) file.

closed 2019-08-19 by martins-mozeiko