lgsvl/simulator #195: Connection with Apollo

dongyy1148, 2019-06-09:
Hi,
I have done some simulations connecting your simulator with Apollo 3.5. Sometimes the simulations run very well, however, sometimes the ego vehicle doesn't follow the path and it runs on a straight line until it meets an obstacle. I want to find out why this kind of problem happens. I have checked my CPU usage, and the simulator only uses about 20%. I also opened channels information using cyber_monitor, but all the channels' frame ratios only change a little when the issue happens. Therefore, I want to ask how can I check the concrete message pass between LGSVL and Apollo. I have read all the python files in the Api folder, but haven't found related commands. Could you tell me how the Apollo pass messages to the simulator and what message it pass and how can I check it? Also the same question for simulator pass message to Apollo. I tried to print out egoControl.throttle, egoControl.steering, egoControl.braking, however, they are all zero when the vehicle follows the Apollo planning. Thanks!

martins-mozeiko, 2019-06-11:
This seems similar to problem we have noticed on few machines.
Check in cyber_monitor topic for control - if you see that lidar_timestamp value not changing then this is the same issue.

The problem is somewhere inside Apollo, but we have not yet found what is wrong. Currently it looks like that control module stops receiving trajectory messages from /apollo/planning. And when it does not receive correct messages it sends wrong commands to simulator - basically accelerate very fast.

If you want to debug then check [ControlComponent::Proc](https://github.com/ApolloAuto/apollo/blob/master/modules/control/control_component.cc#L271) method. If you'll print out `trajectory_msg` variable you'll see that once vehicle looses control values of trajectory_msg message do not change anymore. It only remembers last value it has seen. And because of this control command values, including lidar_timestamp, is set wrong.

dongyy1148, 2019-06-11:
Thanks a lot! I check the simulation again, and the lidar_timestamp doesn't change when the vehicle is out of control.

closed 2019-06-11 by dongyy1148

luqiang21, 2019-06-26:
Hi @dongyy1148 , we rebased our apollo-3.5 fork with official apollo repo. The control problem is not happening in our team. Please update with our latest apollo-3.5 and pull our latest docker image as well.

dongyy1148, 2019-06-26:
@luqiang21 Thanks a lot for the information! How can I update the edition and docker image conveniently?

luqiang21, 2019-06-26:
@dongyy1148 Since apollo changed their history, so when you update our apollo-3.5 repo, you need to reset your `HEAD` to `origin/simulator` after fetch if your remote to our repo is also named `origin`. 
- `git fetch` then `git reset --hard origin/simulator`

For docker image, you just need to do 
`docker pull lgsvl/apollo-3.5`

dongyy1148, 2019-06-26:
@luqiang21 Thank you so much!

luqiang21, 2019-06-26:
@dongyy1148 You're very welcome. Note that when you do `dev_start.sh`, there will be errors about preload, it can be ignored for now since we didn't observe any influence on modules for Apollo. And inside docker, you won't be able to launch `cyber_visualizer`. To launch it, you need to do `unset LD_PRELOAD` inside docker.

dongyy1148, 2019-06-27:
@luqiang21 Got it. Thanks for the info!