lgsvl/simulator #196: Perception in rainy scenario

dongyy1148, 2019-06-10:
System information
    OS Platform and Distribution (Ubuntu 16.04):
    Connected with Apollo 3.5:

I try to change the weather to be rainy and dark to do the simulation, but the Lidar pointcloud is the same as in the no rainy scenario. It seems that rain doesn't influence the perception and path planning.

[image]

martins-mozeiko, 2019-06-11:
That is right. In current implementation lidar is not affected by rain. Currently rain is purely visual effect for cameras.

dongyy1148, 2019-06-11:
Got it. Thanks!

closed 2019-06-11 by dongyy1148

berkerlogoglu, 2019-11-13:
do you plan to add the effect of rain on lidar returns in the upcoming releases?

martins-mozeiko, 2019-11-13:
We would like to have it, but currently we do not have plans for it.