lgsvl/simulator #200: problem with bridge.sh Apollo 3.5, environment variable [CYBER_PATH]

wengchunlei, 2019-06-15:
Thanks LGSVL for great job.
 I am new to lgsvl, installed 3.5 correctly, it downloaded and built without any issues. 

run ./scripts/bootstrap.sh start
->Launched module monitor.
->Launched module dreamview.
->Dreamview is running at http://localhost:8888
and DreamView at http://localhost:8888 show nothing.

run  ./scripts/bridge.sh
->error: [cyber_bridge] Environment variable [CYBER_PATH] not set

Anyone meet the same problem? How can I fix this?
Thank you.

martins-mozeiko, 2019-06-17:
Are you using https://github.com/lgsvl/apollo-3.5 repository for installing Apollo 3.5? Our Cyber Bridge is custom component which does not exist in original Apollo 3.5 repository.

And how are you launching `./scripts/bridge.sh` ? Is it inside same docker container as boostrap is run?

martins-mozeiko, 2019-06-19:
This is probably same issue as #209. Check that you are not running dev_start.sh as root user. Apollo does not like running as root user. Run dev_start.sh under your normal user.

closed 2019-08-19 by martins-mozeiko