lgsvl/simulator #201: About System requirement

yumianhuli2, 2019-06-17:
Recommended system:
    4 GHz Dual core CPU
    Nvidia GTX 1080
    Windows 10 64 Bit

What about this?
    i7-9750H  2.6GHz 
    Nvidia GTX1660Ti
    WIN10 64?

Thank U!

martins-mozeiko, 2019-06-17:
Unity game engine does a lot of job on one thread - main thread. So having higher GHz really helps.
But it all depends what exactly will you be running. If you choose smaller map and don't use all sensors then the simulation will run fine. Only if you'll use big map and enable a lot of sensors the performance may suffer and simulation won't be smooth.

lemketron, 2019-06-17:
Note that the GTX 1080 would be faster than the GTX1660Ti as the GTX1660Ti is more comparable in performance to the GTX 1070 than the GTX 1080.
See https://gpu.userbenchmark.com/Compare/Nvidia-GTX-1660-Ti-vs-Nvidia-GTX-1080/4037vs3603

closed 2019-08-19 by martins-mozeiko