lgsvl/simulator #202: What is the difference between the coordinates of the simulator and Apollo 3.5?

outmanlol, 2019-06-17:
Hey, LGSVL Teams:
        As the title says,What is the difference between the coordinates of the simulator and Apollo 3.5?  Why is the coordinate X Y Z of the lidar on the unity3d simulation platform different from that of apollo35's lidar? Where can I find their transformation relationship?
       Looking forward to your reply.

martins-mozeiko, 2019-06-17:
In Unity for objects the X axis points right, Y is up and Z forward. In Lidar message we are sending points in coordinate system where X is forward, Y is left and Z is up. This seems to what Apollo and Autoware expects.

outmanlol, 2019-06-19:
> In Unity for objects the X axis points right, Y is up and Z forward. In Lidar message we are sending points in coordinate system where X is forward, Y is left and Z is up. This seems to what Apollo and Autoware expects.
@martins-mozeiko Thinks for your reply.  I still have a problem.
In Unity the coordinate of Lidar are 0,2.312,-0.3679201,but in apollo35 are 0,0.9807289,2.312.
I can't get the correct result by using the coordinate transformation you mentioned.
Could you help me?
[image]
[image]

martins-mozeiko, 2019-06-19:
If I'm not mistaken then this extrinsic is between lidar and imu senors. You need to get position of both lidar and imu, then calculate difference to put here. You cannot simply take absolute value of transform of lidar - that will be relative to (0,0,0) position of car.

outmanlol, 2019-06-19:
> If I'm not mistaken then this extrinsic is between lidar and imu senors. You need to get position of both lidar and imu, then calculate difference to put here. You cannot simply take absolute value of transform of lidar - that will be relative to (0,0,0) position of car.

@martins-mozeiko Tanks for your reply. I have get the correct result by using the coordinate transformation you mentioned. By the way ,in apollo for objects the X axis points right, Y is forward and Z   is up. And I have an another question. In Unity ,the Euler angle of Lidar are 0,0,0,the same with imu,But in apollo35 ,the value of four elements of lidar is not 1,0,0,0.

martins-mozeiko, 2019-06-19:
I think 0.707 value for both z and w is 90 degree rotation around vertical axis. I believe for Apollo lidar in normal orientation is looking "to right" not "forwards". That's why it is rotated 90 degrees. In simulator we don't really care about lidar rotation as that is only visual effect.

outmanlol, 2019-06-20:
@martins-mozeiko Thinks for your reply. I have figer out it. I just ignored the coordinates of the other sensors themselves.

closed 2019-06-20 by outmanlol