lgsvl/simulator #204: Get error message from apollo3.5 docker

yumianhuli2, 2019-06-17:
E0617 16:58:16.595429  7072 client.cc:85] [cyber_bridge] Unknown operation received from client (71), disconnecting client

martins-mozeiko, 2019-06-17:
This means you have selected Apollo 3.0 vehicle which is trying to use rosbridge protocol to your specified connection address. You need choose "XE_Rigged-apollo_3_5" which is to use CyberBridge.

closed 2019-06-19 by martins-mozeiko

Avps1, 2019-11-03:
Hi,
So I am getting the same error as mentioned by yumianhuli2, even if I choose ''XE_Rigged-apollo_3_5" in Unity lgsvl . In fact the error starts as soon as I choose the 'Free Roaming' option under LGSVL AUTOMOTIVE SIMULATION(Standalone, Open GL4.5)(See attached screenshot)
[image]

The problem persists even if I choose XE_Rigged-apollo_3_5. 

Error: 

E1103 21:43:18.107486 31124 client.cc:85] [cyber_bridge] Unknown operation received from client (71), disconnecting client

martins-mozeiko, 2019-11-03:
@Avps1 This error can happen only if you have selected vehicle with non CyberRT bridge (for example, rosbridge for Apollo 3.0 or Autoware). Please select 3.5 vehicle.

If you want to debug this further, please show the screenshot how are you selecting Apollo 3.5 vehicle and if this error is still happening then.

Avps1, 2019-11-11:
Hello, 
I ensured that I am running Apollo 3.5 and not 3.0 by checking the Docker Version in DreamView(Snapshot 1: APOLLO_VERSION.png). As you can see in Snapshot2(Unity_Error2.png), I have chosen 'XE Rigged_Apollo_3.5' for Vehicle both in UNity lgsvl and DreamView, but as I stated earlier, the error occurs even before I choose the Vehicle in Unity lgsvl. The error starts being displayed in CLI , as soon as I click 'Free Roaming' in UNity. 

Could you kindly tell me what is the right way to select the Apollo 3.5  vehicle so that this error stops showing?
[image]
[image]

martins-mozeiko, 2019-11-11:
If you see these error messages before choosing vehicle in Free Roaming menu, then that is OK. You can safely ignore these messages. Make sure the error messages stop when you select 3.5 vehicle. After that everything will work correctly.