lgsvl/simulator #206: Car cannot move

Prj1508, 2019-06-18:
Hi, I have bootstrapped apollo and the bridge is connected. I have followed the steps outlined under https://github.com/lgsvl/apollo-3.5, but the vehicle is not able to move. I am using Apollo3.5 on Ubuntu 16.05 with 1070 laptop. Here are snapshots. Please help

[image]

daohu527, 2019-06-18:
As can be seen from the pic, there is no indication that the car is stopped. Normally, Dreamview will display a parking alert in the upper right corner of the grid in front of the car. Have you set up "route editing" in the Dreamview? Please check if the routing path is ready.

Prj1508, 2019-06-18:
Thanks for your response. 
For route editing, all I am doing is following what is detailed as various steps in https://github.com/lgsvl/apollo-3.5 that is
-Navigate to the Route Editing tab.
-Select a destination by clicking on a lane line and clicking Submit Route.
I did not see Submit route specifically, but after clicking the lane line, I clicked on 'Send the routing request". 
Am I supposed to do anything else? 
I tried these steps again, but the result is same. Please let me know

best regards

luqiang21, 2019-06-18:
@Prj1508 It looks like your planning module is not working. Can you check whether your perception module is working correctly? You can enable traffic and check whether Apollo can detect obstacles inside dreamview.

Prj1508, 2019-06-18:
@luqiang21, thanks. Please clarify if you mean to say I should enable traffic besides other sensors and check in dreamview if it can detect those obstacles? This is what I did
1. Bootstrap apollo and establish cyber_bridge
2. Start LGSVL simulator, check enable traffic + enable sensors ( Lidar, Radar, GPS, IMU)
3. In dreamview, under modules, turn on localization module, perception and traffic light module

When I do this, perception and traffic light module automatically gets unselected and I dont see anything in dreamview besides ego vehicle. Here is the snapshot. Please let me know what am I doing wrong
[image]

luqiang21, 2019-06-18:
@Prj1508 Yes, I suggest to enable traffic to check whether your apollo can get obstacles detected. And you don't need to enable `radar` sensor. You need `main camera` and `telephoto camera` enabled as well.
> perception and traffic light module automatically gets unselected  
  
means these modules are crashing. In Apollo 3.5, besides Localization, you need to enable Transform as well to enable perception and Traffic Light.

Prj1508, 2019-06-18:
@luqiang21 . I enabled traffic + main camera and telephoto camera in LGSVL simulator. In dreamview, after I select XE_Rigged_Apollo3.5 and San Fransisco, I see the default screen :+1: 
[image]
I do not see route editing enabled.
Should I enable Localization, Transform, Perception, Traffic Light, Planning, Prediction, Routing in dreamview? 
Also, should I turn on SIm Control in tasks for these to work? 
I am lost in all the possibilities. Please suggest

Prj1508, 2019-06-18:
Also, I enabled all those modules in Dreamview, but route editing is still disabled.

martins-mozeiko, 2019-06-18:
Yes, you need to enable all those other modules you mention. Only then Apollo will localize car and display its location. You do NOT need to turn on SimControl.

Prj1508, 2019-06-19:
Thanks for your inputs. Dreamview is able to see all the obstacles as per the simulator. I am now able to edit and send the route. When I enable control module, it however fails to go to the target point and crashes into an obstacle. Here is the snapshot
[image]
Please let me know what could be wrong? Am I forgetting enabling of any other modules? Thanks

daohu527, 2019-06-19:
@Prj1508 
You should first enable all modules and then set up route editing. It seems that your control module is not enabled and the planning module gets incorrect feedback.
Another possible reason is that the route is not working at all or the location of the car is wrong. You can try again and confirm that the route is started after all modules are opened.
The routing module is to tell the car's navigation line. After that, the planning module will plan a short-term line of 100-200 meters, then send the generated path information to the control module, and then generate the throttle brake steering information to the car, control the car to drive. 
The sensor, the vehicle's dynamics model, and the traffic are all simulated by the simulator, that is, the information is sent to the simulator for interaction.
I think you should look at Apollo's documentation first, and there is a doc about the dreamview will helpful. [here](https://github.com/ApolloAuto/apollo/blob/master/docs/specs/dreamview_usage_table.md)

Prj1508, 2019-06-19:
I enabled all the modules (incl control) before sending route request. The route gets enabled, but the car does not move. When I skip control module, send routing request and then turn on control module, the car moves, but does not stop at the target point and crashes. Here is the screenshot
[image]
I noticed that Radar and GPS modules do not turn on in MKZ standard debug setup mode as well. Please let me know

martins-mozeiko, 2019-06-19:
This is probably same issue as #195. You can read in my comment there where this happens in Apollo 3.5 code.

Btw you do not need to enable Camera and Velodyne modules. These modules are for reading data from real hardware devices. Simulator already is sending lidar/camera messages to Apollo, so there's no need for these modules.

luqiang21, 2019-06-26:
Hi @Prj1508, we rebased our apollo-3.5 fork with official apollo repo. The control problem is not happening in our team. Please update with our latest [apollo-3.5](https://github.com/lgsvl/apollo-3.5) and pull our latest docker image as well.

yanbec, 2019-07-03:
@Prj1508 As for the correct startup procedure you maybe want to have a look at [this video](https://youtu.be/19eo9ZffCKQ)
The project readme of the lgsvl Apollo fork has been updated and now states, as you found out, that you have to set the route before starting the control module to get the car moving.
I ran into the exact same issue regarding the car crashing into obstacles, but as @luqiang21 stated this seems to be resolved.
If you have trouble pulling/cloning the latest Apollo version, have a look at issue https://github.com/lgsvl/simulator/issues/166
Those exact commits solved the problem for me.

martins-mozeiko, 2019-08-19:
Closing as the issue should be fixed in latest apollo-3.5 and apollo-5.0 code. @Prj1508 - if you more issues on this please open new github issue.

closed 2019-08-19 by martins-mozeiko