lgsvl/simulator #207: How to generate protobuf files?

daohu527, 2019-06-18:
I modified the apollo protobuf file. Then I should modify the protobuf file in "lgsvl". I found the file in "lgsvl/simulator/Assets/Scripts/Cyber​​/Protobuf" and I also found that the ".cs" file was generated by tool.  
I use "protobuf-csharp-port" to generate the some file. But the generated file is different from the one in lgsvl. Can you tell me how to generate a prototype file? (Convert to c# with proto2)

daohu527, 2019-06-18:
Maybe "Protobuf-net"?

daviduhm, 2019-06-20:
Yes. We've used [protobuf-net](https://github.com/mgravell/protobuf-net) to generate those .cs files since this library supports protobuf2 and Apollo currently uses protobuf2. Here's the command with flags we've used for converting all proto files to C#: `protogen.exe --csharp_out=cs +names=original +langver=3 -I. **/*.proto`

daohu527, 2019-06-20:
@daviduhm Thx for you reply!
Which version do you use? I used 2.4.0 but found some differences from the current file.

daviduhm, 2019-06-20:
We've used 2.3.16 I believe: 
```
$ protogen.exe --version
protogen 2.3.16+gce9c91fc8
```

daohu527, 2019-06-20:
I use the 2.3.16 version in the path "protogen/bin/Release/net462". But I found the "protobuf-net.dll" size is 269KB but in the simulator is 297KB. The version is:
```
$ protogen.exe --version
protogen 2.3.16
```

Then I think it does't matter. Some file got error like this and others not:   
```
C:\01code\protobuf-net-2.3.16\src\protogen\bin\Release\net462>protogen.exe --proto_path=../net462 --csharp_out=gen +names=original +langver=3 modules/v2x/proto/v2x_service_obu_to_car.proto
modules/perception/proto/perception_obstacle.proto(30,12,30,26): error: type not found: 'common.Point3D'
modules/perception/proto/perception_obstacle.proto(36,12,36,26): error: type not found: 'common.Point3D'
modules/perception/proto/perception_obstacle.proto(48,12,48,26): error: type not found: 'common.Point3D'
modules/perception/proto/perception_obstacle.proto(51,12,51,26): error: type not found: 'common.Point3D'
modules/perception/proto/perception_obstacle.proto(58,12,58,26): error: type not found: 'common.Point3D'
modules/perception/proto/perception_obstacle.proto(86,12,86,26): error: type not found: 'common.Point3D'
modules/perception/proto/perception_obstacle.proto(89,12,89,26): error: type not found: 'common.Point3D'
modules/perception/proto/perception_obstacle.proto(93,12,93,26): error: type not found: 'common.Point3D'
modules/perception/proto/perception_obstacle.proto(157,12,157,25): error: type not found: 'common.Header'
modules/perception/proto/perception_obstacle.proto(158,12,158,28): error: type not found: 'common.ErrorCode'
```
And I compare the file "chassis.cs", seems the ENUM is diff:
<<< lgsvl
```
        [global::ProtoBuf.ProtoContract()]
        public enum DrivingMode
        {
            COMPLETE_MANUAL = 0,
            COMPLETE_AUTO_DRIVE = 1,
            AUTO_STEER_ONLY = 2,
            AUTO_SPEED_ONLY = 3,
            EMERGENCY_MODE = 4,
        }
```
/>>> new generate
```
        public enum DrivingMode
        {
            [global::ProtoBuf.ProtoEnum(Name = @"COMPLETE_MANUAL")]
            CompleteManual = 0,
            [global::ProtoBuf.ProtoEnum(Name = @"COMPLETE_AUTO_DRIVE")]
            CompleteAutoDrive = 1,
            [global::ProtoBuf.ProtoEnum(Name = @"AUTO_STEER_ONLY")]
            AutoSteerOnly = 2,
            [global::ProtoBuf.ProtoEnum(Name = @"AUTO_SPEED_ONLY")]
            AutoSpeedOnly = 3,
            [global::ProtoBuf.ProtoEnum(Name = @"EMERGENCY_MODE")]
            EmergencyMode = 4,
        }
```
Is there some option to control? Or some thing wrong?

I fellow this steps to generate the file.

1. Collect all ".proto" files. Copy to their original path. Such as: 
```
├── canbus
│   └── proto
├── common
│   ├── adapters
│   │   └── proto
│   ├── configs
│   │   └── proto
│   ├── monitor_log
│   │   └── proto
│   ├── proto
│   ├── util
│   │   └── testdata
│   ├── vehicle_model
│   │   └── proto
│   └── vehicle_state
│       └── proto
```
2. Use the ".bat" to generate the cs file. There maybe lots of files, and I use only one for example.
```
protogen.exe --proto_path=../net462 --csharp_out=gen +names=original +langver=3 modules/canbus/proto/chassis.proto
...
```
3. The dir "gen" also need to mkdir the whole path, otherwise it will throw a error which not find the path.

daohu527, 2019-06-21:
Update! The error maybe a issue in protobuf-gen. Must use fully qualified names in proto files to generate c#. [link](https://stackoverflow.com/questions/56695946/can-protobuf-package-be-omitted-when-using/56697113#56697113)  

Then just one problem. The enum code is not generate right.

daviduhm, 2019-06-21:
You're right. We had to fix apollo .proto files and use full namespace to handle those type not found errors. For 'common.Point3D', for example, we've added 'apollo' to it and used 'apollo.common.Point3D' as a workaround.

By the way, we've also fixed [this issue](https://github.com/mgravell/protobuf-net/issues/452) (and I see that you've also commented on this one) for the generator. We are in the process of making a github repo for this protobuf-net so people can use the same library for generating proto cs files as us. In the meantime, it's a simple fix that you just need to add 'f' suffix for float default values. Below is the code snippet of our fix on `src/protobuf-net.Reflection/CSharpCodeGenerator.cs:381`:
```csharp
if (!defaultValue.EndsWith("f"))
{
    defaultValue += "f";
}
```

We are also working on re-generating proto cs files using the latest protobuf-net (2.4.0) since we've recently rebased our internal apollo 3.5 repo to their official master branch and there might be some changes/additions in proto files. Hence, all of the proto cs files will be updated accordingly. Please bear with us on this matter.

daviduhm, 2019-06-22:
@daohu527 Let me ping you for an update that we found a bug in the code snippet above, and here's the updated code on `src/protobuf-net.Reflection/CSharpCodeGenerator.cs:375`:
```csharp
                    switch (defaultValue)
                    {
                        case "inf": defaultValue = "float.PositiveInfinity"; break;
                        case "-inf": defaultValue = "float.NegativeInfinity"; break;
                        case "nan": defaultValue = "float.NaN"; break;
                        default:
                            if (!string.IsNullOrEmpty(defaultValue) &&
                                float.TryParse(defaultValue, out var f) &&
                                !defaultValue.EndsWith("f"))
                            {
                                defaultValue += "f";
                            }
                            break;
                    }
```

And, for the enum, we've confirmed that enums are generated with PascalCase, not CAPITAL_CASE, with the latest protobuf-net. We'll also update our simulator code to use PascalCase enum as well.

daohu527, 2019-06-22:
Good job! You really solve all my problems.

daohu527, 2019-06-26:
Additional, How does the NPC know the lane and the direction? They just move by program or they have AI and know the lane, the direction and the traffic light?

daviduhm, 2019-06-27:
Basically, we have map annotations for each scene, and this map annotations have these information: lane, direction, traffic light, traffic sign, and etc. NPCs know the lane and the direction using the map annotations and drive around following the traffic rules we have. By AI, if you are asking if NPCs are detecting lanes or traffic lights using its own sensors, they are not for performance reasons. Only ego vehicles have sensors like a camera or LiDAR and can be controlled by AD stack.

Let me kindly suggest you to open up a new issue if you have more questions on NPC behaviors or other topics to keep this thread dedicated for protobuf files. Thank you!

closed 2019-07-17 by daohu527