lgsvl/simulator #208: Vehicle can't move and no chassis message

HNwangjiaxing, 2019-06-18:
Hi，I have bootstrapped Apollo and the bridge is connected. Apollo is built under Ubuntu 14.04, and LGSVL is running under Windows. I have followed the steps outlined under https://github.com/lgsvl/apollo，I have set up "route editing" in the Dreamview and enable Localization, Perception, Planning, Prediction, and Control. But the vehicle can't move. Look at the control log. The error is that there is no chassis message. The log is as follows:

[image]

LG-yzz, 2019-06-18:
![316255871420849822](https://user-images.githubusercontent.com/51531852/59688175-b6c46600-920f-11e9-9c58-40c4e46e55fa.jpg
这个是上面的图片

daohu527, 2019-06-18:
Which apollo branch do you use? I can't find the "control.cc" file in apollo 3.5. You can try this [apollo branch](https://github.com/lgsvl/apollo-3.5).
You can also refer to #184

martins-mozeiko, 2019-06-18:
I assume this is with Apollo 3.0.

@HNwangjiaxing What map are you using? SanFrancisco? Does the map load correctly in DreamView and localization works (vehicle position is shown correctly)?

If you open one new terminal and check the chassis topic with `rostopic echo /apollo/canbus/chassis` - does it show something?

closed 2019-08-19 by martins-mozeiko