lgsvl/simulator #209: About CYBER_PATH

yumianhuli2, 2019-06-19:
Hello!
Here is :(already in docker container)
root@in_dev_docker:/apollo/scripts# ./bootstrap.sh 
nohup: appending output to 'nohup.out'
Module monitor is already running - skipping.
Module dreamview is already running - skipping.
root@in_dev_docker:/apollo/scripts# ./bridge.sh 
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0620 09:21:34.858167 20118 global_data.cc:150] [cyber_bridge] host ip: 192.168.1.33
W0620 09:21:34.858201 20118 environment.h:33] [cyber_bridge] Environment variable [CYBER_PATH] not set, fallback t
How to solve it?Thank U!

martins-mozeiko, 2019-06-19:
We have seen that Apollo does not like when it is run as root user. It does not setup some environment variables correctly. Please try running Apollo as non-root user.

yumianhuli2, 2019-06-19:
Yes,I am not the root user for running Apollo docker.So how can I do next?

yumianhuli2, 2019-06-19:
> We have seen that Apollo does not like when it is run as root user. It does not setup some environment variables correctly. Please try running Apollo as non-root user.
Yes,I am not the root user for running Apollo docker.So how can I do next?

martins-mozeiko, 2019-06-19:
I'm not sure I understand when you say that you are "not the root user". Because in log file it clearly says you are root user: `root@in_dev_docker:/apollo/scripts# ./bootstrap.sh`.

Please do not run dev_start.sh with root user or with sudo. Run it under your regular user.

yumianhuli2, 2019-06-20:
> I'm not sure I understand when you say that you are "not the root user". Because in log file it clearly says you are root user: `root@in_dev_docker:/apollo/scripts# ./bootstrap.sh`.
> 
> Please do not run dev_start.sh with root user or with sudo. Run it under your regular user.

If I do not use sudo, just type ./docker/scripts/dev_start.sh,It will call [ERROR] Failed to start docker container "apollo_dev_XXX" based on image:  lgsvl/apollo-3.5:latest!
If I use sudo ./docker/.....dev_start.sh, I got [ok] enjoy,So I need to use sudo.

martins-mozeiko, 2019-06-20:
I think you have not configured your user to allow it to manage docker. Please check docker documentation how to set it up so it docker command can be used as non-root user: https://docs.docker.com/install/linux/linux-postinstall/#manage-docker-as-a-non-root-user

closed 2019-08-20 by martins-mozeiko

jpcloud1577, 2021-01-07:
Hi, I am facing exact same issue even when running bridge.sh as a non-root user:

pm@in_dev_docker:/apollo$ ./scripts/bridge.sh   
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0107 09:20:53.035511 11317 global_data.cc:150] [cyber_bridge] host ip: 192.168.1.134
W0107 09:20:53.035624 11317 environment.h:33] [cyber_bridge] Environment variable [CYBER_PATH] not set, fallback to 

I have followed instructions in the docker official site as suggested by you, but I am still not able to get around this issue. I also tried chown. Please let me know