lgsvl/simulator #211: Debugging Apollo with LGSLV

Prj1508, 2019-06-20:
Hi,

I have apollo 3.5 running with LGSLV and I would like to debug Apollo code base. I followed the steps to initiate GDBServer in VS code for Apollo, but when I run LGSLV, the debugger does not start. The only difference between the two scenarios is turning on SIm control when I am debugging apollo as a standalone. The use case is same otherwise. I have no problems in debugging Apollo in VS Code as a standalone. Pls let me know what am I missing

martins-mozeiko, 2019-06-20:
What do you mean with "debugging apollo as standalone"? Are you running Apollo outside of docker?

You should be able to debug Apollo before launching simulator. Please make sure that works before running simulator. If it does not then this is debugger/docker configuration issue.

Prj1508, 2019-06-21:
I meant debugging in Apollo with only Simcontrol on when I said standalone. I am running Apollo in docker. I can debug all the modules before running simulator, but when I connect simulator, the debugging in VS stops. Please let me know.

Prj1508, 2019-06-21:
These are the steps I am following
1. Bootstrap Apollo
2. Start simulator bridge and sim
3. Open Apollo project in VS code. apply breakpoint as needed (in control module for now) press F5
4. In Dreamview, select the options XE rigged, the modules localization, transform, perception, routing, traffic light, prediction, planning and control. 
5. Send routing request 
When I do these, the debugger crashes

martins-mozeiko, 2019-06-21:
Debugger crashing really sounds like some issue with debugger configuration or apollo docker. As simulator is not really doing anything with debugger itself... I would suggest asking this in Apollo issues, as they are more experienced running debugger with Apollo code.

Prj1508, 2019-07-23:
Hi,
I would like to know the list of messages that are being published from LGSLV to Apollo and how do I monitor them in real time? Please let me know

thanks

martins-mozeiko, 2019-07-23:
You can press F1 when simulator is running to see what topics it subscribed or is publishing.
Then to monitor messages you can use `rostopic echo` for Apollo 3.0 or `cyber_visualizer` for Apollo 3.5 inside Apollo docker container.