lgsvl/simulator #214: vision_ssd_detect not working [Autoware]

orkop465, 2019-06-24:
Hi,
I am attempting to use Autoware alongside LGSVL however I am having some issues with vision_ssd_detect. Whenever I launch vision_ssd_detect either from the launch files or from the runtime manager, I get this error:
[image]
However when running Autoware with LGSVL I can see that the traffic lights are recognized, although the vehicle doesn't take them into account and does not stop for red lights. It does not seem to recognize other vehicles or pedestrians, and attempts to drive straight through them when following the path. Any help would be appreciated.

yanbec, 2019-06-26:
Hi!
Can't exactly help you with your problem, as I'm facing the same with Yolo.
I'll just leave this here for reference, as the "ignoring of information" is probably the same.

https://github.com/lgsvl/simulator/issues/135

orkop465, 2019-06-30:
Hey,
I figured out the issue, the .caffemodel file in 
/home/USER/autoware/docker/generic/data/ssd/models/VGGNet/VOC0712Plus/VGG_VOC0712Plus_SSD_512x512_ft_iter_160000.caffemodel was downloaded incorrectly, this was very obvious due to the fact that it was only 103 bytes. I took the same caffemodel file from another computer with Autoware and that seemed to fix the issue.

closed 2019-06-30 by orkop465

yanbec, 2019-06-30:
@orkop465 did you manage that Autoware now actually takes the obstacle information into account and prevents crashing into other cars?
Would you mind to give me a little more details? Maybe this could help fix my problems with Autoware and Yolo, or at least gets it to work with SSD :)