lgsvl/simulator #216: is there a way to start multi scene within one simulator server ?

ZJLi2013, 2019-06-25:
there is an issue about [running multi robot](https://github.com/lgsvl/simulator/issues/35), ma maybe similar

martins-mozeiko, 2019-06-25:
I'm not sure what you mean by "start multi scene". But if you want to add multiple vehicles in same simulation environment - you can do that right now. Just click + button in menu screen where you add vehicles and choose each vehicle & its connection to bridge. Simulator will create all of them once the map loads.

ZJLi2013, 2019-06-26:
I actually want to make a parallel running of LG simulator.  so one  way, I suppose, is to start one LG simulator server, and connect a few clients(on each client node), and each client to load one scene. 

is it possible ? 

Thanks

martins-mozeiko, 2019-07-12:
This is currently not possible. Also it would require at lot of resources. Loading scene and simulating multiple clients costs a lot of performance/memory on one machine. It is better to run separate simulations on different machines.

But we are working on new simulator version that later will allow launching simulator instances on different clusters. Basically you will be able to invoke multiple simulations on different machines from one central control page.

closed 2020-04-07 by EricBoiseLGSVL