lgsvl/simulator #217: Get distance to lane center in simulator

n1try, 2019-06-26:
I'd like to know whether it is possible to either directly get the go vehicle's distance from its current lane's center or, alternatively, the lane boundaries in local- or world coordinates.

PS: Thanks for this project and all your effort! It's going to help me a lot with my M.Sc. thesis.

martins-mozeiko, 2019-06-26:
This depends on what you mean by "current lane". Because simulator does not care about what EGO vehicle is doing - is it following lane or driving in straight lines, or just crashing into building. It only responds to commands received from AD stack. It does not know what decision AD stack is doing (following lane or just accelerating straight).

And if you control EGO vehicle from AD software such as Autoware or Apollo, then you have vector map information in AD stack. You can load vector map and get whatever information you need from this map with AD stack specific functionality.

If you want to just get distance to closest lane, then you can check how `follow_closest_lane` Python API function is implemented for NPC vehicles. Its code is in [NPCControllerComponent.cs#L1639](https://github.com/lgsvl/simulator/blob/91d60e9232ebfc2b3fc013265af8ef02c1d907b0/Assets/Scripts/NPCControllerComponent.cs#L1639) There you can see call to `MapManager.Instance.GetClosestLane`  which gets closest line segment (list of points) in world coordinates. Then there is a loop that finds part of segment that is closest to specified position.

n1try, 2019-06-26:
Thanks for your reply! The second part is what I was looking for.
It'd be very helpful if the simulator exposed a method like `GetClosestLane()` that could be called remotely. Is there any plans to integrate such? Otherwise I could probably do it by myself, but it would take some time for me to get into the code.

EricBoiseLGSVL, 2020-04-07:
@muety this would be best accomplished with API and a custom sensor.

closed 2020-04-07 by EricBoiseLGSVL