lgsvl/simulator #218: Bibtex citation  for LGSVL ?

madhurbehl, 2019-06-26:
Is there a preferred bibtex citation available for the LGSVL sim to include in a research paper ? Count find one on the website or the git page. 
We are using the simulator for our research and would like to credit the authors/dev properly, rather than just a link to the url.

bshin-lge, 2019-07-02:
Hello madburdehl,
We do plan to publish a paper on the simulator project later which you'll be able to cite, but for now you can probably just fill in a misc entry with our Github project information.

closed 2020-04-07 by EricBoiseLGSVL

luqiang21, 2020-06-11:
Hi @madhurbehl,
we recently published paper of the simulator [LGSVL Simulator: A High Fidelity Simulator for Autonomous Driving](https://arxiv.org/abs/2005.03778), it will be presented at 2020 ITSC. Feel free to cite it.

lemketron, 2021-03-24:
For anyone searching for citation information for the SVL Simulator, please use the following:
```
@ARTICLE{2020arXiv200503778R,
    author = {{Rong}, Guodong and {Shin}, Byung Hyun and {Tabatabaee}, Hadi and
     {Lu}, Qiang and {Lemke}, Steve and
     {Mo{\v{z}}eiko}, M{\={a}}rti{\c{n}}{\v{s}} and {Boise}, Eric and
     {Uhm}, Geehoon and {Gerow}, Mark and {Mehta}, Shalin and
     {Agafonov}, Eugene and {Kim}, Tae Hyung and {Sterner}, Eric and
     {Ushiroda}, Keunhae and {Reyes}, Michael and {Zelenkovsky}, Dmitry and
     {Kim}, Seonman},
    title = "{SVL Simulator: A High Fidelity Simulator for Autonomous Driving}",
    journal = {arXiv e-prints},
    keywords = {Computer Science - Robotics, Computer Science - Machine Learning, Electrical Engineering and Systems Science - Systems and Control},
    year = 2020,
    month = may,
    eid = {arXiv:2005.03778},
    pages = {arXiv:2005.03778},
    archivePrefix = {arXiv},
    eprint = {2005.03778},
    primaryClass = {cs.RO}
}
```