lgsvl/simulator #219: Is it possible to integrate new sensors like a sonar sensor?

alexsimon1, 2019-06-27:
Dear LGSVL team,

Is it possible to integrate new sensors like a sonar sensor?
What would be the steps to accomplish this?

We want to recreate a Mercedes Benz with several
Sonar / radar / lidar-sensors and cameras.

Best regards

Alexander Simon

EricBoiseLGSVL, 2019-07-08:
@alexsimon1 

Hello.  Yes, it is very possible to have a sonar sensor.  I recommend using the radar sensor as guide.  I would use a mesh for the directivity.  This can be done in code or with modeling software.  Objects that enter, stay or exit using Unity api calls can collect objects in range.  After the sonar is collecting the data correctly, look at the bridge data types and writer methods, to send.

Hope this helps.

alexsimon1, 2019-07-08:
Thank you for your help

closed 2019-07-08 by alexsimon1

lemketron, 2019-07-08:
@alexsimon1 
It would be great if you would consider sending us a pull request once you have this working so we can add it to the LGSVL Simulator as a new sensor type for others to use... Thanks!