lgsvl/simulator #220: ModuleNotFoundError: No module named 'lgsvl'

dongyy1148, 2019-06-28:
Hi LGSVL team,

I used to use python file to do the simulation connected with Apollo, and it works well. However, my system got down a few days ago, and I believe some setting may change. Now, when I run python files again it returns that ModuleNotFoundError: No module named 'lgsvl' even though I did source command. Could you help look at it? Thanks!
[image]
[image]

luqiang21, 2019-06-28:
@dongyy1148 You can try to reinstall the package again following instructions [here](https://github.com/lgsvl/simulator/tree/master/Api)

dongyy1148, 2019-07-01:
@luqiang21 Thanks! It works!

closed 2019-07-01 by dongyy1148