lgsvl/simulator #221: Error building Apollo fork rosbridge for Windows

abhishek-nilugal, 2019-06-30:
Hi,

I'm trying to build rosbridge for Apollow fork in Windows. I noticed that the CmakeLists has been developed for linux and it therefore points to wrong file when trying to build for Windows. Despite correcting that error and running `catkin_make`, I get the following error:

```
C:\Users\saiab\OneDrive\Desktop\LGSVL\apollo\ros_pkgs>catkin_make
Base path: C:\Users\saiab\OneDrive\Desktop\LGSVL\apollo\ros_pkgs
Source space: C:\Users\saiab\OneDrive\Desktop\LGSVL\apollo\ros_pkgs\src
Build space: C:\Users\saiab\OneDrive\Desktop\LGSVL\apollo\ros_pkgs\build
Devel space: C:\Users\saiab\OneDrive\Desktop\LGSVL\apollo\ros_pkgs\devel
Install space: C:\Users\saiab\OneDrive\Desktop\LGSVL\apollo\ros_pkgs\install
####
#### Running command: "cmake C:\Users\saiab\OneDrive\Desktop\LGSVL\apollo\ros_pkgs\src -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCATKIN_DEVEL_PREFIX=C:\Users\saiab\OneDrive\Desktop\LGSVL\apollo\ros_pkgs\devel -DCMAKE_INSTALL_PREFIX=C:\Users\saiab\OneDrive\Desktop\LGSVL\apollo\ros_pkgs\install -G NMake Makefiles" in "C:\Users\saiab\OneDrive\Desktop\LGSVL\apollo\ros_pkgs\build"
####
CMake Error at CMakeLists.txt:1:
  Parse error.  Expected a command name, got unquoted argument with text
  "C:\opt\ros\melodic\x64\share\catkin\cmake\toplevel.cmake".
```

I'm neither familiar with Catkin nor with Cmake; It is therefore taking infinitely long to debug this. Can you please help me?

yanbec, 2019-06-30:
Hi!
It would probably be helpful if you could include your CMakeLists.txt as the error seems to be in there.

abhishek-nilugal, 2019-06-30:
This is how my CMakeLists looks like at the moment:
```
# ROS Bridge
cmake_minimum_required(VERSION 2.8.3)
include(/opt/ros/melodic/x64/share/catkin/cmake/toplevel.cmake)
```

Is it correct? Should I make any changes?

I was able to build partly after making this modification. But I'm stuck with yet another another error as below:

```
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(54): note: Siehe Verweis auf die Klasse Vorlage-Instanziierung "boost::arg<9>", die kompiliert wird.
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(53): note: Siehe Verweis auf die Klasse Vorlage-Instanziierung "boost::arg<8>", die kompiliert wird.
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(52): note: Siehe Verweis auf die Klasse Vorlage-Instanziierung "boost::arg<7>", die kompiliert wird.
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(51): note: Siehe Verweis auf die Klasse Vorlage-Instanziierung "boost::arg<6>", die kompiliert wird.
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(50): note: Siehe Verweis auf die Klasse Vorlage-Instanziierung "boost::arg<5>", die kompiliert wird.
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(49): note: Siehe Verweis auf die Klasse Vorlage-Instanziierung "boost::arg<4>", die kompiliert wird.
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(48): note: Siehe Verweis auf die Klasse Vorlage-Instanziierung "boost::arg<3>", die kompiliert wird.
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(47): note: Siehe Verweis auf die Klasse Vorlage-Instanziierung "boost::arg<2>", die kompiliert wird.
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(46): note: Siehe Verweis auf die Klasse Vorlage-Instanziierung "boost::arg<1>", die kompiliert wird.
[ 51%] Linking CXX executable C:\Users\saiab\OneDrive\Desktop\LGSVL\apollo\ros_pkgs\devel\lib\rosauth\ros_mac_authentication.exe
LINK: command "C:\PROGRA~2\MICROS~1\2017\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\Hostx64\x64\link.exe /nologo @CMakeFiles\ros_mac_authentication.dir\objects1.rsp /out:C:\Users\saiab\OneDrive\Desktop\LGSVL\apollo\ros_pkgs\devel\lib\rosauth\ros_mac_authentication.exe /implib:C:\Users\saiab\OneDrive\Desktop\LGSVL\apollo\ros_pkgs\devel\lib\ros_mac_authentication.lib /pdb:C:\Users\saiab\OneDrive\Desktop\LGSVL\apollo\ros_pkgs\devel\lib\rosauth\ros_mac_authentication.pdb /version:0.0 /machine:x64 /INCREMENTAL:NO /subsystem:console C:\opt\ros\melodic\x64\lib\roscpp.lib C:\opt\rosdeps\x64\lib\boost_filesystem-vc141-mt-x64-1_66.lib C:\opt\ros\melodic\x64\lib\rosconsole.lib C:\opt\ros\melodic\x64\lib\rosconsole_log4cxx.lib C:\opt\ros\melodic\x64\lib\rosconsole_backend_interface.lib C:\opt\rosdeps\x64\lib\log4cxx.lib C:\opt\rosdeps\x64\lib\boost_regex-vc141-mt-x64-1_66.lib C:\opt\ros\melodic\x64\lib\roscpp_serialization.lib C:\opt\ros\melodic\x64\lib\xmlrpcpp.lib C:\opt\ros\melodic\x64\lib\rostime.lib C:\opt\ros\melodic\x64\lib\cpp_common.lib C:\opt\rosdeps\x64\lib\console_bridge.lib C:\opt\rosdeps\x64\lib\boost_system-vc141-mt-x64-1_66.lib C:\opt\rosdeps\x64\lib\boost_thread-vc141-mt-x64-1_66.lib C:\opt\rosdeps\x64\lib\boost_chrono-vc141-mt-x64-1_66.lib C:\opt\rosdeps\x64\lib\boost_date_time-vc141-mt-x64-1_66.lib C:\opt\rosdeps\x64\lib\boost_atomic-vc141-mt-x64-1_66.lib crypto.lib kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib /MANIFEST /MANIFESTFILE:C:\Users\saiab\OneDrive\Desktop\LGSVL\apollo\ros_pkgs\devel\lib\rosauth\ros_mac_authentication.exe.manifest" failed (exit code 1181) with the following output:
LINK : fatal error LNK1181: Eingabedatei "crypto.lib" kann nicht ge÷ffnet werden.
NMAKE : fatal error U1077: "C:\opt\rosdeps\x64\bin\cmake.exe": Rückgabe-Code "0xffffffff"
Stop.
NMAKE : fatal error U1077: ""C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\nmake.exe"": Rückgabe-Code "0x2"
Stop.
NMAKE : fatal error U1077: ""C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\nmake.exe"": Rückgabe-Code "0x2"
Stop.
Invoking "nmake" failed
```

Would be glad if you can help me!

martins-mozeiko, 2019-07-12:
If I'm not mistaken ROS1 does not officially support Windows. Have you managed to build whole ROS1 on Windows? Note that you need to build Apollo ROS1 modifications, not vanilla ROS1 as they have modified message wire format (protobuf). Does Apollo ROS1 builds on Windows without our rosbridge addon?

abhishek-nilugal, 2019-07-13:
Hi @martins-mozeiko , 
ROS does support windows (http://wiki.ros.org/Installation/Windows). I was able to build ROS, however not the ROS Bridge.

Thank very much for your inputs! I however had to abandon this effort due to change of priorities. I hope to see the support for Windows sometime soon.

closed 2020-04-07 by EricBoiseLGSVL